yamcs/yamcs #446: Need to allow arrays larger than 10000 elements

merose, 2020-09-22:
We have an instrument that sends back a spectrograph that is currently modeled as an <ArrayParameterType> with 8-bit elements. Our instrument returns spectra of up to about 17000 entries. We've run into a Yamcs limitation which seems somewhat arbitrary.

In SequenceEntryProcessor.java, line 33:

    static final int MAX_ARRAY_SIZE = 10000;

and lines 125–128:

        if (ts > MAX_ARRAY_SIZE) {
            throw new XtceProcessingException("Resulted size of the array " + pe.getParameter()
                    + " exceeds the max allowed: " + ts + " > " + MAX_ARRAY_SIZE);
        }

so we get an error when Yamcs is unpacking that parameter from the container:

    12:17:22.911 _global [19] XtceTmExtractor got exception in tmextractor :
        org.yamcs.xtceproc.XtceProcessingException: Resulted size of the array ParaName: data
        paraType:ArrayParameterType name:dataBig numberOfDimensions:1 aliases: {} exceeds the max
        allowed: 16302 > 10000

If there needs to be a hard limit, it would be nice if it were a configurable value.

xpromache, 2020-12-04:
Just looking back at this issue: a binary parameter type is more efficient than an array of bytes. 

But I agree we should have an option for that 10000.

merose, 2020-12-04:
In our particular case it is an array of single-precision floating-point values that represent a spectrum. The client will process the values to produce a spectral plot. Storing them as a binary parameter value, while more efficient, would be less convenient because the data typing is lost. However, I'll work with the instrument team to see if that would be a reasonable workaround.

xpromache, 2020-12-04:
Are those values received into one single packet?
Currently we cannot extract arrays spawning multiple packets.

One thing I would do if increasing the limit would be to create specialized
types to store more efficiently array of primitives.
Currently an array is simply Value[] where Value is a full blown java
object which means it occupies  3-4 times more memory than if we had a
float[].

It's all doable in a rainy Belgian afternoon...

commit 76466f12 (2021-03-31): array improvements

- allow to extract arrays from packets even without using the ArrayParameterEntry - this is possible because in XTCE 1.2 array types can specifiy their size
- added config validation for processor config
- allow to configure maxArraySize (fixes #446)
closed 2021-03-31 by xpromache