yamcs/yamcs #447: Parameters calculated by an algorithm do not appear to be stored in the parameter archive

merose, 2020-09-24:
I have three parameters, two of which arrive in CCSDS packets through a data link, and a third which is calculated from the other two in a configured `<MathAlgorithm>`. When I use the Get Parameter Samples API on the three, I get values from the parameter archive for the first two, but for the third I appear only to get values from the in-memory cache.

These two calls return data for the entire range based on the desired number of samples, as you can see:

    $ curl -s 'http://localhost:8090/api/archive/viper/parameters/ViperRover/NssIo/detector2DiscriminatorLSB/samples?start=2020-09-24T00:22:10.463Z&stop=2020-09-24T00:52:10.463Z&count=894&order=asc' | grep -c '"time"'
    865
    $ curl -s 'http://localhost:8090/api/archive/viper/parameters/ViperRover/NssIo/detector2DiscriminatorMSB/samples?start=2020-09-24T00:22:10.463Z&stop=2020-09-24T00:52:10.463Z&count=894&order=asc' | grep -c '"time"'
    865

and the times cover most of the range, as for `detector2DiscriminatorLSB` here:

    $ curl -s 'http://localhost:8090/api/archive/viper/parameters/ViperRover/NssIo/detector2DiscriminatorLSB/samples?start=2020-09-24T00:22:10.463Z&stop=2020-09-24T00:52:10.463Z&count=894&order=asc' | grep '"time"' | head -n 1
        "time": "2020-09-24T00:23:10.853Z",
    $ curl -s 'http://localhost:8090/api/archive/viper/parameters/ViperRover/NssIo/detector2DiscriminatorLSB/samples?start=2020-09-24T00:22:10.463Z&stop=2020-09-24T00:52:10.463Z&count=894&order=asc' | grep '"time"' | tail -n 1
        "time": "2020-09-24T00:52:10.085Z",

However, when I change the parameter name to the computed parameter, I get a warning in the log:

    17:57:49.773 _global [40] SingleParameterArchiveRetrieval No parameter id found in the parameter archive for /ViperRover/NssIo/detector2Filtered

and I also do not get back the desired number of samples. Instead, I only get samples for a few minutes:

    $ curl -s 'http://localhost:8090/api/archive/viper/parameters/ViperRover/NssIo/detector2Filtered/samples?start=2020-09-24T00:22:10.463Z&stop=2020-09-24T00:52:10.463Z&count=894&order=asc' | grep -c '"time"'
    343

with the times:

    $ curl -s 'http://localhost:8090/api/archive/viper/parameters/ViperRover/NssIo/detector2Filtered/samples?start=2020-09-24T00:22:10.463Z&stop=2020-09-24T00:52:10.463Z&count=894&order=asc' | grep '"time"' | head -n 1
        "time": "2020-09-24T00:46:45.992Z",
    $ curl -s 'http://localhost:8090/api/archive/viper/parameters/ViperRover/NssIo/detector2Filtered/samples?start=2020-09-24T00:22:10.463Z&stop=2020-09-24T00:52:10.463Z&count=894&order=asc' | grep '"time"' | tail -n 1
        "time": "2020-09-24T00:52:10.085Z",

I must have something misconfigured. How to I ensure that computed parameters values make it in to the parameter archive so I can retrieve them from a client?

merose, 2020-09-24:
I should have clarified: the parameter calculation in the `<MathAlgorithm>` is triggered by a change in one of the other parameters, so the number of values generated or received for each of the three parameters should be the same, I would think.

xpromache, 2020-09-24:
Do you have the AlgorithmManager in the processor.yaml -> ParameterArchive?
ParameterArchive:
 services:
...
   - class: org.yamcs.algorithms.AlgorithmManager

This is responsible for re-running the algorithms when the ParameterArchive
is built from the raw packets.

merose, 2020-09-24:
Yes, we do AlgorithmManager configured. Our current processor.yaml matches what is in the `quickstart` project. Is there a better example we should be looking at?

But you've given me a clue. I'll debug a little and see when AlgorithmManager is called.

merose, 2020-09-24:
Hmmm... when I compare with yamcs/examples/simulator/src/main/yamcs/etc/processor.yaml, I see that the AlgorithmManager is configured for the Archive and ParameterArchive processors, too. That's missing in `quickstart`. I'll add those configurations to ours and see if it solves the problem.

merose, 2020-09-24:
Yes, that was it. Will close this issue. Thank you for the help. Will compare more carefully with the other example configurations. You might want to update the `quickstart` configuration to include the AlgorithmManager service for the other processors.

closed 2020-09-24 by merose