yamcs/yamcs #449: Is there any support for dynamically-sized arrays or binary data?

merose, 2020-09-25:
XTCE supports dynamic array or binary data sizes by looking up the length from a parameter. Does Yamcs support this in any way?

    <ParameterTypeSet>
      <IntegerParameterType name="ImageLength">
        <IntegerDataEncoding sizeInBits="32" byteOrder="mostSignificantByteFirst" encoding="unsigned" />
      </IntegerParameterType>
      <BinaryParameterType name="ImageData">
        <BinaryDataEncoding>
          <SizeInBits>
            <DynamicValue>
              <ParameterInstanceRef parameterRef="imageLength"/>
            </DynamicValue>
          </SizeInBits>
        </BinaryDataEncoding>
      </BinaryParameterType>
    </ParameterTypeSet>
    <ParameterSet>
      <Parameter parameterTypeRef="ImageLength" name="imageLength" />
      <Parameter parameterTypeRef="ImageData" name="imageData" />
    </ParameterSet>

merose, 2020-09-26:
I see we can probably work around this by using a `<StringDataType>` that is length-prefixed, since we only need a byte array to hold image data. I also see that the spreadsheet MDB loader supports length-prefixed binary data. One thought I had would be to allow the parameter reference with a `<DynamicValue>` size, but check after the container is constructed that the reference is actually a length prefix, and convert the SequenceEntry list accordingly, to match what might have been loaded from the spreadsheet loader. That would be more restrictive than the full XTCE form, but would be workable.

xpromache, 2020-09-26:
That should work, please give it a try and it doesn't we will fix it...

merose, 2021-04-12:
I have submitted a pull request, https://github.com/yamcs/yamcs/pull/549, for supporting XTCE dynamic length references for binary data encodings of parameter and argument types. As I noted in the PR, this may be incomplete because I may have missed uses of the data encodings that I am not familiar with. However, I did include sample XTCE and unit tests, so perhaps it may serve as a start.

xpromache, 2021-04-12:
Thanks for the pull request, I will have a look at it towards the end of the week.

There is another way to do that (similar with what is done from the spreadsheet), see:
https://github.com/yamcs/yamcs/blob/master/yamcs-core/src/test/resources/xtce/ref-xtce.xml

It uses algorithms with an artificial size set to -1 so perhaps not very standard.

I'm not sure how your version is supposed to work for commands - does the user need to input the size in addition to the binary argument? What if the size of the binary doesn't match with the size argument ?

merose, 2021-04-12:
Yes, will look at that as well. I found the custom encoders/decoders while I was learning the code.

The version I submitted requires an argument to the command specifying the encoding size. That was the only way I could see how to accomplish the dynamic length using standard XTCE. If the size does not match the size of the binary argument, the argument value is either truncated or padded, as for fixed-length encoding.

xpromache, 2021-04-20:
merged into master.

closed 2021-04-20 by xpromache

xpromache, 2021-04-27:
Pushed one more change allowing the DynamicIntegerValue size to be specified with a slope such that the resulting size in the packet can be in bytes rather than bits.
For example:

```
      <BinaryArgumentType name="BinaryType">
        <BinaryDataEncoding>
          <SizeInBits>
            <DynamicValue>
              <ArgumentInstanceRef argumentRef="size"/>
              <LinearAdjustment slope="8"/>
            </DynamicValue>
          </SizeInBits>
        </BinaryDataEncoding>
      </BinaryArgumentType>