yamcs/yamcs #454: Cannot reference an AggregateArgumentType from within an Argument on an ArgumentList

lorenzo-gomez-windhover, 2020-10-01:
Hi there!

Thanks so much for yamcs! 

In the `TelemetryMetaData` object  one can do this:

```

<ParameterTypeSet>
                <AggregateParameterType name="MM_HkPacket_t">
                    <MemberList>
                        <Member name="Address" typeRef="BaseType/uint32_LE"/>
                        <Member name="BytesProcessed" typeRef="BaseType/uint32_LE"/>
                        <Member name="CmdCounter" typeRef="BaseType/uint8_LE"/>
                        <Member name="DataValue" typeRef="BaseType/uint32_LE"/>
                        <Member name="ErrCounter" typeRef="BaseType/uint8_LE"/>
                        <Member name="FileName[0]" typeRef="BaseType/int8_LE"/>
                        <Member name="FileName[1]" typeRef="BaseType/int8_LE"/>
                        <Member name="FileName[2]" typeRef="BaseType/int8_LE"/>
                        <Member name="FileName[3]" typeRef="BaseType/int8_LE"/>
                        <Member name="FileName[4]" typeRef="BaseType/int8_LE"/>
                        <Member name="FileName[5]" typeRef="BaseType/int8_LE"/>
                        <Member name="FileName[6]" typeRef="BaseType/int8_LE"/>
                        <Member name="FileName[7]" typeRef="BaseType/int8_LE"/>
                        <Member name="FileName[8]" typeRef="BaseType/int8_LE"/>
                        <Member name="FileName[9]" typeRef="BaseType/int8_LE"/>
                        <Member name="FileName[10]" typeRef="BaseType/int8_LE"/>
                        <Member name="FileName[11]" typeRef="BaseType/int8_LE"/>
                        <Member name="FileName[12]" typeRef="BaseType/int8_LE"/>
                        <Member name="FileName[13]" typeRef="BaseType/int8_LE"/>
                        <Member name="FileName[14]" typeRef="BaseType/int8_LE"/>
                        <Member name="FileName[15]" typeRef="BaseType/int8_LE"/>
                        <Member name="FileName[16]" typeRef="BaseType/int8_LE"/>
                        <Member name="FileName[17]" typeRef="BaseType/int8_LE"/>
                        <Member name="FileName[18]" typeRef="BaseType/int8_LE"/>
                        <Member name="FileName[19]" typeRef="BaseType/int8_LE"/>
                        <Member name="FileName[20]" typeRef="BaseType/int8_LE"/>
                        <Member name="FileName[21]" typeRef="BaseType/int8_LE"/>
                        <Member name="FileName[22]" typeRef="BaseType/int8_LE"/>
                        <Member name="FileName[23]" typeRef="BaseType/int8_LE"/>
                        <Member name="FileName[24]" typeRef="BaseType/int8_LE"/>
                        <Member name="FileName[25]" typeRef="BaseType/int8_LE"/>
                        <Member name="FileName[26]" typeRef="BaseType/int8_LE"/>
                        <Member name="FileName[27]" typeRef="BaseType/int8_LE"/>
                        <Member name="FileName[28]" typeRef="BaseType/int8_LE"/>
                        <Member name="FileName[29]" typeRef="BaseType/int8_LE"/>
                        <Member name="FileName[30]" typeRef="BaseType/int8_LE"/>
                        <Member name="FileName[31]" typeRef="BaseType/int8_LE"/>
                        <Member name="FileName[32]" typeRef="BaseType/int8_LE"/>
                        <Member name="FileName[33]" typeRef="BaseType/int8_LE"/>
                        <Member name="FileName[34]" typeRef="BaseType/int8_LE"/>
                        <Member name="FileName[35]" typeRef="BaseType/int8_LE"/>
                        <Member name="FileName[36]" typeRef="BaseType/int8_LE"/>
                        <Member name="FileName[37]" typeRef="BaseType/int8_LE"/>
                        <Member name="FileName[38]" typeRef="BaseType/int8_LE"/>
                        <Member name="FileName[39]" typeRef="BaseType/int8_LE"/>
                        <Member name="FileName[40]" typeRef="BaseType/int8_LE"/>
                        <Member name="FileName[41]" typeRef="BaseType/int8_LE"/>
                        <Member name="FileName[42]" typeRef="BaseType/int8_LE"/>
                        <Member name="FileName[43]" typeRef="BaseType/int8_LE"/>
                        <Member name="FileName[44]" typeRef="BaseType/int8_LE"/>
                        <Member name="FileName[45]" typeRef="BaseType/int8_LE"/>
                        <Member name="FileName[46]" typeRef="BaseType/int8_LE"/>
                        <Member name="FileName[47]" typeRef="BaseType/int8_LE"/>
                        <Member name="FileName[48]" typeRef="BaseType/int8_LE"/>
                        <Member name="FileName[49]" typeRef="BaseType/int8_LE"/>
                        <Member name="FileName[50]" typeRef="BaseType/int8_LE"/>
                        <Member name="FileName[51]" typeRef="BaseType/int8_LE"/>
                        <Member name="FileName[52]" typeRef="BaseType/int8_LE"/>
                        <Member name="FileName[53]" typeRef="BaseType/int8_LE"/>
[... 69 lines cut ...]
                infob.setRangeMin(iat.getValidRange().getMinInclusive());
                infob.setRangeMax(iat.getValidRange().getMaxInclusive());
            }
        } else if (argumentType instanceof FloatArgumentType) {
            FloatArgumentType fat = (FloatArgumentType) argumentType;
            if (fat.getValidRange() != null) {
                infob.setRangeMin(fat.getValidRange().getMin());
                infob.setRangeMax(fat.getValidRange().getMax());
            }
        } else if (argumentType instanceof EnumeratedArgumentType) {
            EnumeratedArgumentType eat = (EnumeratedArgumentType) argumentType;
            for (ValueEnumeration xtceValue : eat.getValueEnumerationList()) {
                infob.addEnumValue(toEnumValue(xtceValue));
            }
        }
        return infob.build();
    }
```

A workaround of course is to list out every member of the struct in the `ArgumentList`; essentially flattening the struct. However, it'd be nice to just reference an `AggregateType` in commands directly.

Is there a particular reason why this happens? 
Is it not supported in the standard?

Thanks in advance.

merose, 2020-10-01:
Aggregate types are not yet supported in Yamcs for argument types in commands, according to the Yamcs server manual: https://yamcs.org/docs/yamcs-server-manual/mdb/loaders/xtce/. See the Compatibility > Commands section at the bottom of that page.

lorenzo-gomez-windhover, 2020-10-01:
I see it! Thanks!

xpromache, 2020-10-02:
Adding support for this is not such a big issue (but as always in Yamcs we don't implement things nobody uses) so I pushed now some changes in the cmd-aggregate branch. I'll do some more tests and merge it to master.

lorenzo-gomez-windhover, 2020-10-02:
Awesome! Thanks!

closed 2020-10-15 by lorenzo-gomez-windhover