yamcs/yamcs #455: Yamcs 5.1.3 UDP Datalink / TM Port blocking

nmaas87, 2020-10-06:
Dear all,
I upgraded my existing YAMCS Server instance from 4.10.9 to 5.1.3 and am currently using it with version  v1.4.0.rc1.
I did have a test installation of 5.0, however, I did not use it long enough to find any errors (it was just a quick install and test - but I made the jump to the latest and greatest when 5.1.3 arrived 👍🏻 ) - however, I found some very troubleing problem: It looks like if my payload has not been sending TM for some time, the TM Link "dies".

To explain this further: I am using Yamcs Server with an UDP TM Datalink as well as an UDP TC Datalink. 
These datalinks are fed / read by some Python Scripts, i.e. that means TMs are sent from Python via UDP to the UDP TM Datalink.
For TCs, another Python Script provides an UDP Server which awaits the UDP TC packets from Yamcs.

If the payload goes silent for some time and then starts talking again - it happens from time to time that I see that data on my python scripts being routed through TM to Yamcs - but I never see anything arriving in Yamcs... I can still send TCs from Yacms which also trigger normal actions in the payload (everything fine there), but no TMs are incoming anymore. If I disable the UDP TM Datalink in i.e. Yamcs Studio - and then enable it again, immediately new TMs arrive - as if there never has been a problem. But all data generated between the Payload start sending data - and disabling/enabling the UDP TM Datalink is lost.

So as I made the jump from 4.10.9 to 5.1.3 I cannot where this problem was introduced. 4.10.9 had a rock-solid Datalink and I never had any issue with that, would probably have had that running for weeks without end - if it weren't for system level upgrades and reboots of the VM ;).

I guess it does not matter which Yamcs Studio I use, this seems to be a particular problem of the server and I am sorry that I cannot provide more explicit input, but I hope it somewhat helps to maybe finding a problem 👍🏻 

Thanks a lot,

Nico

xpromache, 2020-10-06:
Hi Nico,

Thanks for the report.

Can you quantify "some time"? How long do you need to wait for the problem
to happen?
I guess you don't notice anything in the logs?

I tried to reproduce on the quickstart project which also uses a UDP link +
a python script and could not immediately (but I only waited for around 12
minutes). Can you please give it a try:
https://github.com/yamcs/quickstart

Is the python script running on the same machine?
If not, could there be that some network restart or firewall cause the
problem?

I cannot see much change with those links in between the two Yamcs versions
so I don't know what could go wrong.

nicolae

nmaas87, 2020-10-06:
Hi @xpromache - thanks for the answer,

I cannot quantify, as I usually had to have the payload working all the time for checkouts and if I disabled TM for sometime, commanded it to send again and saw this error happening, I did just do my workaround, as I had to get results for tha payload.

So I would need to see if I can create a time-window in which I can take a closer look at that problem, sorry to disappoint.

The Python Script is running on another VM, but connection between both systems is solid - the only thing that changed was the upgrade from Yamcs 4.10.9 to 5.1.3, all other scripts were not touched or changed. But thanks already for taking a look 👍🏻

nmaas87, 2020-10-06:
Ok I tried it again today several times... but I could not reproduce the problem, which is very bad. I know I had it several times (in the region of 5-7 times) within 4 workdays, but I currently cannot reproduce or give any furter inputs... I would leave this open and see if I can add something if it reappears...

nmaas87, 2020-10-14:
I really had not much time to test this issue, however I will close it for now. I will reopen if it happens again or reference to it. Thanks a lot!

closed 2020-10-14 by nmaas87