yamcs/yamcs #457: CCSDS max packet size as option / configureable

nmaas87, 2020-10-08:
https://github.com/yamcs/yamcs/blob/a58f39d8d4e9a28f566f052e271d163e76b8c619/yamcs-core/src/main/java/org/yamcs/tctm/CcsdsPacket.java#L10

https://github.com/yamcs/yamcs/blob/a58f39d8d4e9a28f566f052e271d163e76b8c619/yamcs-core/src/main/java/org/yamcs/tctm/TmFileReader.java#L68

nmaas87, 2020-10-08:
( I posted to fast, so now there is even some text in the issue, instead of only some position in text )

xpromache, 2020-10-13:
The TmFileReader class is very specific to the USOC/ISS setup, and should
be gone from Yamcs core;  it expects packets in very specific formats.

Data links such as FilePollingTmDataLink should be rewritten to use the
configurable packet preprocessors like the other data links.

nmaas87, 2020-10-14:
Thanks for your answer - this is however specific to the TmFileReader class. I understand we can generate our own preprocessors to create own data links.

However, is it possible to make the parameter MAX_CCSDS_SIZE as a default to 1500 - but configurable via YAML so that we could adopt it other payload needs without having to recompile the whole server ourselves? I guess that FilePollingTmDataLink should be adable with less hassle or am I wrong there? Thanks a lot for your time and effort 👍🏻

xpromache, 2020-10-19:
TmFileReader is used from FilePollingTmDataLink and FileTmPacketProvider. I understand your use it in connection to FilePollingTmDataLink, right?

I think I will get rid of the FileTmPacketProvider (can be replaced with a python script sending TM from file) and make the FilePollingTmDataLink use the packetInputStream similar with what TcpTmDataLink does.

I will rename the TmFileReader into a UsocPacketInputStream - the peculiarity of this is that it can read packets in multiple formats (raw CCSDS, PaCTS and HRDP but the last two are not relevant for users outside ISS/Columbus) with auto detection of the format. On the other hand it makes some assumptions about the CCSDS packets which are not necessary correct outside the Columbus/ISS world. Ideally it should move to our yamcs-usoc package.

Then I can add the option for the CCSDS packet length on the UsocPacketInputStream but you can also use the other CcsdsPacketInputStream or even the GenericPacketInputStream which can parse other types of packets.

xpromache, 2020-10-19:
I pushed the changes above in the branch tm-file-reader-refactor, I suggest you give it a try.
I will merge it to the master one of these days.

nmaas87, 2020-10-20:
Hey Nicolae,

thank you very much for your hard work. It makes sense to bundle all USOC/ISS specific links at one place and give the generic ones more flexiblity. I think that will make things a lot easier to implement without needing to modify the "core" Yamcs. Thank you :)

closed 2020-10-20 by nmaas87