yamcs/yamcs #458: HTTP API "Get Parameter Value" doesn't report expired parameters as expired

leighgarbs, 2020-10-08:
I've defined an expiration time on a SequenceContainer in my XTCE telemetry definition using "DefaultRateInStream", like so:
```xml
<SequenceContainer name="SomeContainer" shortDescription="Contains some pretty important stuff">
    <DefaultRateInStream basis="perSecond" minimumValue="1" />

    ... more definition goes here ...

</SequenceContainer>
```
My understanding is that this leads Yamcs to expect parameters included in this container to be updated at least once a second.

When I stop my telemetry, the web GUI does show the parameters in this container as expired after roughly a second (it draws a little clock next to their rows in the Parameter viewer), and an active Python API parameter subscription for one of these parameters will generate a callback message with a "validity_status" of "EXPIRED" when a parameter expires, if it's configured to.  However, querying for the value of these an expired parameter [as described here](https://yamcs.org/docs/yamcs-http-api/processing/get-parameter-value/) returns an "acquisitionStatus" of "ACQUIRED":
```json
{
  "id": {
    "name": "someParameter",
    "namespace": "/CoolSpaceSystem"
  },
  "rawValue": {
    "type": "UINT32",
    "uint32Value": 0
  },
  "engValue": {
    "type": "ENUMERATED",
    "stringValue": "NIFTY",
    "sint64Value": "0"
  },
  "acquisitionTime": "2020-10-08T00:13:28.330Z",
  "generationTime": "1970-01-12T14:03:55.290Z",
  "acquisitionStatus": "ACQUIRED",
  "processingStatus": true,
  "acquisitionTimeUTC": "2020-10-08T00:13:28.330Z",
  "generationTimeUTC": "1970-01-12T14:03:55.290Z",
  "expireMillis": "1900"
}
```
I would have expected this query to return an acquisitionStatus of EXPIRED.  Am I missing something?

xpromache, 2020-10-09:
I don't think you are missing anything, it is indeed a questionable
behaviour. Originally we never sent expired parameters but was expecting
the client to compute it based on the expireMillis property.

Later we added an option on the subscription to receive a copy of the
parameter with the expired status set if the parameter has expired. This is
done on the client facing component of Yamcs and it relieves the clients
from doing this computation themselves.

However the parameter cache which is used when you retrieve the value with
the get, is never updated. I think the reason for that was:
- the expiration status of parameters depends when you look at them - i.e.
if you get a value now and look at it after 3 seconds then it is expired no
matter what the status says. This is the reason I considered originally
this to be a client side business.
- it was not very obvious how to implement it in the cache making sure that
the expired parameter update will not overwrite the new values received.
Not impossible but I preferred avoiding running into concurrency issues.

We also have to consider that the cache remembers older values as well.
When you look at an  older value (i.e. not at the last value of a
parameter) - should this be considered expired or not?

I think an easy fix would be to perform the same checking when sending
parameters as response to the get method. This is probably relatively
straightforward to implement.

A more fundamental fix would be to change them in the cache (if we consider
the cache as storing the parameters in the state they were in realtime) but
as I said that is more tricky and I wouldn't hold my breath for it.

leighgarbs, 2020-10-17:
Is "acquisitionStatus" really a property that should be associated with parameter values?  Maybe it's something that should instead be associated with a parameter.  The "INVALID" acquisitionStatus seems like the only setting that makes sense to apply to a parameter value in isolation.  The other settings only seem appropriate for a sequence or stream of parameter values.

akhenry, 2023-01-23:
We are hitting this issue in Open MCT as well. We rely on subscriptions to get the initial state of a parameter using the `sendFromCache` option, but there are cases where we rely on [Get Parameter Value](https://docs.yamcs.org/yamcs-http-api/processing/get-parameter-value/) and [Batch Get Parameter Values](https://docs.yamcs.org/yamcs-http-api/processing/batch-get-parameter-values/) to establish the initial state of a parameter. Right now these two paths give us inconsistent acquisition information.

I could understand this behavior if Get Parameter Value returned no `acquisitionStatus` at all. Right now though it is returning an incorrect `acquisitionStatus`, which is worse than not returning it at all. To be clear though, I'd prefer it returned correct acquisition information :)