yamcs/yamcs #462: Should be able to specify destination entityId when sending CFDP file

AdamElliott-AB, 2020-10-20:


xpromache, 2020-10-20:
It is possible to specify the destination entity id (key _destinationId_ ) in the configuration of the CFDP service:

```
services:
...
  - class: org.yamcs.cfdp.CfdpService
    args:
     sourceId: 5
     destinationId: 24
     sequenceNrLength: 4
     maxPduSize: 512
     incomingBucket: "cfdpDown"
     eofAckTimeout: 3000
     maxEofResendAttempts: 5
```
Or do you mean to specify it for each file transfer individually when starting the transfer (in case there are multiple destinations)?

AdamElliott-AB, 2020-10-20:
Yes, when there is more than one destination, I want to be able to specify it when starting a transfer.

xpromache, 2020-10-20:
Unfortunately this feature is not planned to be implemented in the immediate future because the only project we use CFDP for does not need it
If you really need it you have to either implement it yourself and make a pull request or contact my company to ask some official support.

AdamElliott-AB, 2020-10-20:
Understood

xpromache, 2020-12-07:
Finally we need this feature so it will be part of the next release.

closed 2020-12-07 by xpromache