yamcs/yamcs #464: Comment in comment in command stack is not exported to XML

merose, 2020-10-27:
When adding a command to a command stack you can add a comment. However, the comment is neither showed in the UI of the stack nor is it exported to the XML form. Both would be nice.

(Other capabilities of stacks that would be nice: Ability to edit a command in a stack. (Currently can delete a command and re-add it, dragging to the right position. A bit clumsy.) Ability to invoke a command stack. Ability to set relative timing of commands when invoking a stack (i.e., execute a sequence of commands, without requiring a user click). Ability to check telemetry values before invoking a command in a stack.)

fqqb, 2020-10-28:
There's supposed to be comment support, but there indeed seems to be a problem with it when adding it from the dialog. I'll fix that.

I agree with the other list. Some of these we've been wanting to add for a while already, but it just takes a while to get to it. Specific to the automatic execution: when we add this, it'd probably be rather based on acknowledgments: the next command only executes after a specific (configurable) acknowledgment of the previous command checks out.

> Ability to check telemetry values before invoking a command in a stack.

This already should be possible: the MDB allows the definition of arbitrary transmission constraints for commands.

fqqb, 2021-01-29:
Closing because OT as well as most of the other suggestions are good now on master.

Exception is the 'relative timing', which I want to wait out a while until the implementation of the current feature set stabilises. It is however now possible to execute the remainder of a stack all at once (all are queued in order).

closed 2021-01-29 by fqqb