yamcs/yamcs #467: Downloaded Paramter data is incorrect

AdamElliott-AB, 2020-11-16:
Parameter data downloaded from Telemetry->Parameters->"Historical Data" tab->"Download Data" is incorrect.

See https://github.com/AdamElliott-AB/quickstart for a minimal example demonstrating this issue.

Steps to reproduce:
1. build/run example
2. Simulated data for the "Measurement" parameter is received correctly (values from 0-60): http://localhost:8090/telemetry/parameters/%2Fmyproject%2FMeasurement/chart?c=myproject__realtime
3. Download data from the "Historical Data" tab for the "Measurement" parameter and compare with actual data: http://localhost:8090/telemetry/parameters/%2Fmyproject%2FMeasurement/data?c=myproject__realtime

xpromache, 2020-11-17:
Thanks for the report.

Looks like you have uncovered a bug in the telemetry processing. When you download data from the web, because the parameter archive has not been consolidated, Yamcs starts a "replay" but only with the selected parameters. 

Replay means that it extracts the packets from the archive and the parameters from the packets.

To optimize the retrieval, it does not extract all the parameters but only the necessary ones and there is an algorithm to decide which are the necessary ones (in addition to those requested by the user). In this case, you have the hierarchy
```
CCSDSPacket      
   - CCSDS_Packet_ID   
   - CCSDS_Packet_Sequence  
    - CCSDS_Packet_Length  
   -> TelemetryPacket
        <no parameters inside> 
       -> Spacecraft2 
            - Measurement
```

The XTCE allows to specify the position of a parameter as relative to the previous entry (default) or absolute from the beginning of the container. In your XTCE file, Measurement position is relative to the previous entry.

So practically to extract "Measurement" , it needs first to extract the three parameters from CCSDSPacket.

However the bug is that when it sees that Measurement  is the first parameter in Spacecraft2, it looks in the TelemetryPacket (as the parent of Spacecraft2), it doesn't find any parameters there and it stops. It should look higher in the hierarchy but instead it ends up extracting the Measurement at the wrong offset in the packet.

This problem does not appear in the realtime retrieval or the parameter archive consolidation because there it extracts all the parameters so it doesn't use this algorithm to limit the retrieval. It would not appear either if you had a parameter defined in TelemetryPacket which explains why we haven't encountered the bug until now.

We will fix the bug for the next release. 

In the mean time, you can force the offset of Measurement by defining it like this in the XTCE:
```
<SequenceContainer name="Spacecraft2">
                <EntryList>
                    <ParameterRefEntry parameterRef="Measurement" >
                       <LocationInContainerInBits referenceLocation="containerStart">
                           <FixedValue>48</FixedValue>
                       </LocationInContainerInBits>
                     </ParameterRefEntry>
                </EntryList>
```
Another way to fix it is to add in the processor.yaml, "subscribeAll: true" to the config of the ArchiveRetrieval processor type.
This option will disable the optimization algorithm I was mentioning above.
```
ArchiveRetrieval:
  services:
    - class: org.yamcs.tctm.ReplayService
    - class: org.yamcs.algorithms.AlgorithmManager
  config:
    subscribeAll: true
    parameterCache:
      enabled: false
```

AdamElliott-AB, 2020-11-17:
Thank you for looking at and addressing this issue.

closed 2020-11-17 by xpromache