yamcs/yamcs #471: XtceAssembler does not export algorithms from XtceDb

Blimes, 2020-12-03:
The XtceAssembler class export parameters and commands very well in xml format but I cannot see algorithms (syntetic parameters) when I back to XtceDb through the XtcedbFactory, even in the generated xml. Will XtceAssembler class exports also algorithms to xml in future release ?

xpromache, 2020-12-03:
I think it can be done but those XTCE algorithms are not very standard.

Do you want to import it to another system? I have a hard time thinking it will work just like that...

Blimes, 2020-12-03:
Yes I'd like to export the XtceDB object to standard xtce xml file and than import to another system that will factory the XtceDB (with algorithms). From the Xtce standard I saw that in the TelemetryMetaDataType support algorithms in the AlgorithmSet; Could be a possibile solution to include the algorithms in the generated xtce xml file as customization of the default supported AlgorithmType of the XTCE standard ?

xpromache, 2020-12-03:
May I know on which system you want to import the data exported from Yamcs?

I don't think producing a valid XML will be very difficult but I have great
doubts that the algorithms exported will work on your target system.

Blimes, 2020-12-03:
Sure, I need to export XTCE from Yamcs to a proprietary software bases on SCOS2k-like mission databases. 
Algorithms are produced by SCOS system and read  by the yamcs-scos2k plugin.

xpromache, 2020-12-03:
The yamcs-scos2k plugin loads the algorithms from SCOS and sets their
language to SCOS2K-OL and the text to the OL expression.
So you want to get that OL expression exported into the algorithm text
field  in the XTCE xml file ?

What I don't understand: if your proprietary software is based on SCOS2K
MIB - then why do you need to convert the MIB to XTCE using Yamcs?

In any case, I will look at adding the export of the algorithms in XTCE one
of these days.

Blimes, 2020-12-03:
Yes the aim is to export OL Expressions of algorithms to the XTCE XML file, allowing reload mdb from the xtce xml file. By this way i will use only the XtceDbLoader from xml because the xml file is more portable than the entire scos2k files.

thank you.

xpromache, 2020-12-07:
I pushed support for this in the master branch of Yamcs. I have updated also the yamcs-scos2k to latest Yamcs (there were quite some changes because at some point we changed those XTCE objects to use builders in order to implement type inheritance).

Please give it a try if you can and let me know if you have problems.

I would also appreciate if you could share your experience with Yamcs compared to your other proprietary system, we are always looking for ways to improve... (you can send that privately if you want).

closed 2020-12-07 by xpromache