yamcs/yamcs #474: Question: Kubernetes?

JoshSharpe, 2020-12-09:
Hi,

I hate asking questions through GitHub tickets but didn't see a more effective way to ask questions.

Has anyone set-up their ground systems and yamcs services using Kubernetes? I was curious about what they found and if was worth it based on the current implementation of yamcs and the replication, underlying data handling, etc.

xpromache, 2020-12-09:
Hi Josh, we also have an old fashioned mailing list: https://groups.google.com/g/yamcs/
But don't worry, here is fine as well.

I don't think currently Kubernetes it's a good target for Yamcs. It could make more sense if we replaced the embedded storage engine (RocksDB)  with a server based one. I always thought this will be done one day (we try to keep the storage engine as an independent component) but it's not an insignificant amount of work. 

In the past when I tried to use PostgreSQL I had problems with the relatively big latency when accessing the database but nowadays I think it could work with some smart caching strategy.

JoshSharpe, 2020-12-11:
Thanks! I'll be sure to subscribe.

I agree and assumed the same. It seemed like it would be fairly tricky so I'm glad I asked if others had considered before diving too deeply into it.

I'll keep an eye our for that change to be initiated and try to help! If there's enough demand it seems like a worthwhile and educational endeavor.

closed 2020-12-11 by JoshSharpe