yamcs/yamcs #475: Nested Command/Control

nmaas87, 2020-12-14:
Hi there,

I have a new feature request and while I was thinking about some smaller parts, I made my mind up and I think this concept will create a more wholesome and complete experience. 

The basic idea was to have the usual Tcp/Udp TM Datalinks as a kind of "upsteam" port, meaning they should mirror  the incoming CCSDS pakets and forward them to another client via TCP or UDP so that this client could use it (i.e. an EGSE to visualize flight data). But thinking about this alone, it would make sense to make the output also filterable, meaning that this datalink would only forward data from certain APIDs, PacketType and PacketSubtypes. If one would implement something like this, it would give the possibility to spread out certain subsystems/Level 2 Payloads of a Level 1 Payload to other Yamcs instances and allow them to just see their data. And as this is basically a streteched out command & control strucutre for different payload responsibilities, it only makes sense to also include a TC uplink port, which also should have the possibility to filter and only allow certain command types in (APID/PacketType/PacketSubType). Also, for security reasons, commands via this TC uplink port should show up in a special command queue which could be then handled like the normal queues in Yamcs to "just let everything through" or acknowledge each command. This would also be nice for training of new operations personell and class 1 Payload Facilitiy Responsible Centers could give their class 2 Payload colleagues easy access to just their system parts via their connection.

A short image to describe this idea a bit better

[image]

Kind regards

Nico

xpromache, 2020-12-14:
For TM transfer between two Yamcs servers, we have a stream replication service which I think is better than just a UDP connection because if the cascaded server is disconnected for a while, the replication will recover the missing data.

The replication service copies all the data passing in one stream but you can create individual streams for the cascaded servers with StreamSQL instructions.
For the EGSE connection, a custom TCP service is required to send whatever data the EGSE requires, so not sure this can be generic (depends on the particular EGSE software). 

For commanding, the situation is more complicated though. If you want the cascaded server to put commands in the upstream server queue, it better use the standard HTTP interface which allows to authenticate users. Maybe you don't want to trust the connecting clients to provide binary packets because there could be anything inside, even commands not authorized. I think the best  is for the interface to send commands in "engineering" form (e.g. command name + argument values).

This cascaded commanding service may be useful to have as a generic component.

Note however that APID/PacketType/PacketSubType are specific to Columbus/ISS environment, Yamcs does not know anything about it. Command restrictions in Yamcs are imposed on command names; the command definition in MDB can include project specific restrictions on APID, PacketType,etc.

nmaas87, 2021-09-24:
Thanks for the information 👍🏻 
I don't actually know why I said APID back then, probably because it transported the idea.
The actual usage would be for a multi-user-multi-experiment faclity (think IceCubes) - except non-ISS :)
Gonna probably solve the issue by writing my own external CCSDS router and attach it to the multiple Yamcs Instances via TCP TM/TC links.
Thanks!

closed 2021-09-24 by nmaas87