yamcs/yamcs #484: Is the dot notation for *AggregateTypes Supported in YAMCS?

lorenzo-gomez-windhover, 2021-01-11:
Hi everyone,

Hope you are all doing well.

I attempted the following in our XTCE file:

```
                <SequenceContainer name="CFE_EVS_EVENT_MSG_MID">
                    <EntryList>
                        <ParameterRefEntry parameterRef="CFE_EVS_Packet_t.Payload"/>
                    </EntryList>
                    <BaseContainer containerRef="cfs-ccsds/cfs-tlm-hdr">
                        <RestrictionCriteria>
                            <ComparisonList>
                                <Comparison parameterRef="cfs-ccsds/ccsds-apid" value="8"/>
                            </ComparisonList>
                        </RestrictionCriteria>
                    </BaseContainer>
                </SequenceContainer>
```

The part of interest is `<ParameterRefEntry parameterRef="CFE_EVS_Packet_t.Payload"/>`.  YAMCS errors out when this is attempted with the following error:

```
14:14:22.491 _global [1] YamcsServer Failure while attempting to validate configuration: org.yamcs.xtce.DatabaseLoadException: Cannot resolve reference SpaceSystem: cfe_evs name: CFE_EVS_Packet_t.Payload type: PARAMETER

```

Here is the definition of `CFE_EVS_Packet_t` (Parameter and Type):

```
<AggregateParameterType name="CFE_EVS_Packet_t">
    <MemberList>
        <Member name="Payload" typeRef="CFE_EVS_Packet_Payload_t"/>
    </MemberList>
</AggregateParameterType>
....
 <Parameter name="CFE_EVS_Packet_t" parameterTypeRef="CFE_EVS_Packet_t"></Parameter>
```

The xtce document[ V1.2](https://www.omg.org/spec/XTCE) says this about some symbols such as ".":

```
Note on Names
Parameter, and MetaCommand and other major entity names within this database may be any length but may only
contain numeric, a-z letters, underscores, hyphens, or backslashes. The characters ‘/’, ‘.’, ‘[‘, ‘]’ and ‘:’ are expressly
reserved. The ‘/’ is used as the SpaceSystem separator (Unix and HTTP style). The ‘:’ is reserved for future use as a
selector for data from other SpaceSystems. The ‘.’ is used to select members of aggregate Parameters and Argurments.
The square brackets are reserved for array indexes. Names are case sensitive.
```

I can always flatten out the structure, so this is no deal-breaker. But I'm just curious (perhaps I'm doing something wrong?).

Thanks in advance.

xpromache, 2021-01-11:
Good catch.

This file:
https://github.com/yamcs/yamcs/blob/master/yamcs-core/src/test/resources/xtce/BogusSAT-2.xml
is using '/' instead of '.' (for example on line 543).

In the XTCE 1.1 it was not specified properly and that xml file has been
written by one of the XTCE authors (I copied it from here
https://gitlab.com/dovereem/xtcetools), that's why I implemented it like
that.

I will fix it in the next version; until then please use "/".

lorenzo-gomez-windhover, 2021-01-11:
Awesome, thanks!

commit da7c80c9 (2021-01-12): Use '.' for references to aggregate members.

Fixes #484
closed 2021-01-12 by xpromache