yamcs/yamcs #485: Nested AggregateTypes in Containers

lorenzo-gomez-windhover, 2021-01-12:
Hi there.

This is a follow-up on #484 

Is the following supposed to be supported by YAMCS at the moment?

```
<EntryList>
    <ParameterRefEntry parameterRef="CFE_EVS_Packet_t"></ParameterRefEntry>
    <xtce:ParameterRefEntry parameterRef="CFE_EVS_Packet_t.Payload"></xtce:ParameterRefEntry>
    <xtce:ParameterRefEntry parameterRef="CFE_EVS_Packet_t.Payload.Spare2"></xtce:ParameterRefEntry>
</EntryList>
```

```
<AggregateParameterType name="CFE_EVS_Packet_Payload_t">
    <MemberList>
        <Member name="Spare2" typeRef="BaseType/uint8_LE"/>
    </MemberList>
</AggregateParameterType>
<AggregateParameterType name="CFE_EVS_Packet_t">
    <MemberList>
        <Member name="Payload" typeRef="CFE_EVS_Packet_Payload_t"></Member>
    </MemberList>
</AggregateParameterType>
```

YAMCS errors out when parsing `CFE_EVS_Packet_t.Payload.Spare2`:
`13:16:37.844 _global [1] YamcsServer Failure while attempting to validate configuration: org.yamcs.xtce.DatabaseLoadException: Cannot resolve reference SpaceSystem: cfe_evs name: CFE_EVS_Packet_t.Payload.PacketID type: PARAMETER
	org.yamcs.xtceproc.XtceDbFactory.resolveReferences(XtceDbFactory.java:246)`

Sorry for the lengthy issue, I hope my question makes sense.

xpromache, 2021-01-12:
No, the second and third entries in the EntryList are invalid according to me.

Aggregate parameters have to be included in full, you cannot include only a member.

xpromache, 2021-01-12:
Sorry, I now re-read the issue #484 and see that it was the same thing actually (I didn't read properly yesterday).

So no, you cannot include parts of a parameter in a packet. It's like you have a 32 bits parameter but include only 5 bits in a packet - the parameter only makes sense if the 32 bits are present.
I guess we could instantiate the other members with some default value if set but currently this is not implemented.

You can reference those aggregate members in conditions because once the full parameter is extracted then you can put conditions on members. So the problem I fixed yesterday refers to how you reference those aggregate members in conditions, not in an entry list.

lorenzo-gomez-windhover, 2021-01-13:
Thanks so much for your help :+1:

closed 2021-01-13 by lorenzo-gomez-windhover