yamcs/yamcs #492: Byte Order in Parameters

lorenzo-gomez-windhover, 2021-01-19:
Hi there,

I was reading the documentation and found the following:

```
Yamcs does not currently support the XTCE way of describing byte ordering for parameter encoding.
```
 on this [page](https://docs.yamcs.org/yamcs-server-manual/mdb/container-definitions/#little-endian-parameter-encoding).

The way we define telemetry and telecommanding in YAMCS is through auto-generated XTCE. The way it is implemented at the moment is that Aggregation is implemented through AggregateTypes. So byte_order, byte_size, etc is defined on the ParameterTypes/ArgumentTypes children of our XTCE.

My question is: is what the documentation says still true? If we found ourselves running flight software such as CFS on Big Endian hardware and it was defined correctly on the XTCE, will YAMCS not be able to parse the telemetry correctly because it is defined at the ParameterType level?

Hope my questions make sense.

Thank you in advance,
Lorenzo

xpromache, 2021-01-19:
That statement is not entirely correct and should be fixed. Before 1.2 XTCE allowed to specify byte order in a complicated manner. Yamcs only supports big and little endian.
XTCE 1.2 added the little/big endian specification in addition to the complicated one they were using previously. So now Yamcs supports XTCE 1.2 big/little endian definition but not the arbitrary order definition.

This is the relevant excerpt from the XTCE 1.2 specs:

```xml
  <simpleType name="ByteOrderType">
                <annotation>
                        <documentation xml:lang="en">Describe a byte order: big/little or byte list.</documentation>
                </annotation>
                <union memberTypes="xtce:ByteOrderCommonType xtce:ByteOrderArbitraryType"/>
        </simpleType>
        <simpleType name="ByteOrderCommonType">
                <annotation>
                        <documentation xml:lang="en">Common byte orderings: most significant byte first (also known as big endian) and least significant byte first (also known as little endian).</documentation>
                </annotation>
                <restriction base="string">
                        <enumeration value="mostSignificantByteFirst"/>
                        <enumeration value="leastSignificantByteFirst"/>
                </restriction>
        </simpleType>
        <simpleType name="ByteOrderArbitraryType">
                <annotation>
                        <documentation xml:lang="en">Describe a byte order using a byte list. The list is viewed as representing memory, the first item in the list is address 0. 
For mostSignificantByteFirst/big endian, the high order byte is the first byte in the list and has the highest significance followed by the less significant bytes ending with the least significant byte. 
For leastSignificantByteFirst/little endian, the first byte starts with the least significant byte which is first in the least and ends at the highest significant byte. 
For example given the value 0x0A0B0C0D the following example orderings can be formed. For mostSignificantByteFirst/big endian the significances would be listed as 3 (0x0A), 2 (0x0B), 1 (0x0C), 0 (0x0D) with ‘3’ being first in the list,
 and for leastSignificantByteFirst/little endian as 0 (0x0D), 1 (0x0C), 2 (0x0B), 3 (0x0A) with ‘0’ being first in the list. See DataEncodingType.</documentation>
                </annotation>
                <restriction base="string">
                        <pattern value="(0|1|2|3|4|5|6|7|8|9|10|11|12|13|14|15)(,(0|1|2|3|4|5|6|7|8|9|10|11|12|13|14|15))*"/>
                </restriction>
        </simpleType>
```

xpromache, 2021-01-19:
Actually if the question is whether a C struct can be mapped one to one to an aggregate parameter type, the answer is that it always depends on how the compiler organizes the memory. The C structs may have gaps if the compiler needs to align the members at word boundaries.
Yamcs aggregates don't have gaps.

gcc has some options to control how the structs are represented in memory:
https://gcc.gnu.org/onlinedocs/gcc-4.4.4/gcc/Structure_002dPacking-Pragmas.html

lorenzo-gomez-windhover, 2021-01-19:
No need to worry about that. We have that part cover. Our XTCE gets generated from data that comes straight from DWARF compiled elf files that get used for flight software ;). We know everything about our binaries; padding; byte_ordering, etc.

xpromache, 2021-02-01:
Closing as I think the support for the little/big endian order is clarified. If more exotic byte or bit orders are required we do have the possibility to plug into an algorithm that reads the raw value out of the packet.

closed 2021-02-01 by xpromache