yamcs/yamcs #493: Rendering Nested Aggregate Types in YAMCS WebApp

lorenzo-gomez-windhover, 2021-01-20:
Hi there,

hope you are all doing well.

Not sure if you guys are aware of this, but it looks like the webapp does not construct the parameter path that gets sent to YAMCS server correctly when it is an AggregateType inside of another Aggregate Type:

[image]

As you can see when I click on `AppInfo`, which is an Aggregate Type inside of the `Payload` AggreggateType, it gives the following error:

```
 157: Responding '404 Not Found': Nonexistent array/aggregate path in name cfs/cfe_es/CFE_ES_OneAppTlm_t.AppInfo
```

As you can tell it makes sense it failed; `cfs/cfe_es/CFE_ES_OneAppTlm_t.AppInfo` is incorrect. It should be `cfs/cfe_es/CFE_ES_OneAppTlm_t.Payload.AppInfo`

I tried my best to fix this myself on webapp, but honestly couldn't do it due to my lack of experience with the APIs you guys use. It looks like it has something to do with the following snippet of code on `ParameterSummaryTab.ts`:

```

  changeParameter(qualifiedName: string) {
    this.yamcs.yamcsClient.getParameter(this.yamcs.instance!, qualifiedName).then(parameter => {
      this.parameter$.next(parameter);

      if (this.parameterValueSubscription) {
        this.parameterValueSubscription.cancel();
      }
      this.parameterValueSubscription = this.yamcs.yamcsClient.createParameterSubscription({
        instance: this.yamcs.instance!,
        processor: this.yamcs.processor!,
        id: [{ name: qualifiedName }],
        abortOnInvalid: false,
        sendFromCache: true,
        updateOnExpiration: true,
        action: 'REPLACE',
      }, data => {
        this.parameterValue$.next(data.values ? data.values[0] : null);
      });
    });
  }
```

The `qualifiedName` value is incorrect at this point; it is `cfs/cfe_es/CFE_ES_OneAppTlm_t.AppInfo`, which is  the incorrect path.

Hope I explained the issue clear enough.

Any ideas on how to fix this?

Thanks
Lorenzo

fqqb, 2021-01-20:
Indeed, nested aggregates and arrays are a bit tricky API-wise due to the type being shared for all members. And this particular page was never tested with it. I'll investigate a fix.

lorenzo-gomez-windhover, 2021-01-22:
Tested the new changes. This is awesome! Thanks :+1: ! Will wait till it is merged into `master` to close issue.

fqqb, 2021-01-23:
Thanks for verifying. We usually close if resolved on master (it is now). If there's any further bug, do say, and we'll reopen/fix.

closed 2021-01-23 by fqqb