yamcs/yamcs #498: A way to store user data into alarms

nikhilmandlik, 2021-01-27:
I am currently trying to store some user data into alarms but I could not find any successful way of doing so.

eg scenario:  In a given system with 5 total users and 5 Tags, there are currently 2 active alarms and admin wants to assign alarms to individual users to monitor those alarms and add tags to them.

I have tried following,
using [edit alarms documentation ]( https://docs.yamcs.org/yamcs-http-api/alarms/edit-alarm/) to edit alarm with  `"state": "unacknowledged"` but no luck, getting error as filed in issue #497 

Please suggest any other way to How can we achieve this?

xpromache, 2021-01-27:
Currently the only possibility is to add a comment (string) to the acknowledge or cleared calls.

nikhilmandlik, 2021-01-27:
@xpromache thanks for quick reply.
I was wondering if I can use [buckets](https://docs.yamcs.org/yamcs-http-api/buckets/) to solve this.
Can I store user and alarms mapping there? Any thoughts?

xpromache, 2021-01-27:
The functionality we have now relates to triggered alarms (i.e. a parameter that went out of limits). I think you are looking to define additional  information to the alarm itself - for example to a parameter  which can potentially go out of limits not to a specific instance of the out of limits event.

Currently the parameter limits / alarms are loaded from the XTCE mission database, I guess the info you want to add could somehow fit there.

But then  you want it dynamic - this gets tricky - what if you add some dynamic parameters/tags to an alarm which disappears from a new version of the XTCE mission database?

Requires some thinking...

You could implement of course some application specific storage in buckets but I believe an important subject like this warrants better server support.

nikhilmandlik, 2021-02-01:
hi @xpromache, I do have a quick question regarding how Alarms acknowledge works.
I have noticed that any acknowledged alarms are getting auto cleared for some reason, is this expected behavior? 
Is this configurable?

[image]

xpromache, 2021-02-01:
The alarm which is acknowledged and also the processOK = true (parameter is back within limits) clears automatically unless the latching flag is set. Unfortunately currently there is no way to set the latching flag.

There is another non configurable flag: autoAcknowledge  - if this is set, the alarm will be auto-acknowledged as soon as the processOK = true. 

The handling of alarms is inspired from the ISA 18.2 Alarm Management standard but the XTCE definition is more simplistic. 
I thought of configuring the latching and the autoAck via the XTCE AncillaryDataSet properties of the BaseAlarmType but never got to do it.
If you need it, I can give it a try, does not sound like a lot of work.

A bit longer term but related to the subject of this issue is to have dynamic alarm definition data outside of the XTCE. There we could define (override) the XTCE alarms and also configure the alarms generated by events which are not possible to configure right now.

xpromache, 2021-02-01:
Here is a state diagram (copied and adapted from the above-mentioned standard) which should hopefully make it clearer how the alarms work:
https://drive.google.com/file/d/1B0XKGrL8r2JYJvuSpOEHbvpn0EGpWuwD/view?usp=sharing

I will add it somewhere in the manual.