yamcs/yamcs #499: Is it possible to make web displays scrollable?

lorenzo-gomez-windhover, 2021-01-28:
Hello everyone,

I came across the following issue when rendering YAMCS studio display on YAMCS Webapp Displays utility:

[image]

As you can see on the left on YAMCS Studio, I am able to scroll all the way down. However on the webapp, I'm not able to.

Any chance I'm missing something?

Or is this capability not added yet?

Thanks in advance
Lorenzo

fqqb, 2021-01-29:
I agree this is undesirable. It's not going to be a quick-fix, because I want to bundle this work when visiting related topics like zoom-to-fit and zoom in/out. The interaction between the canvas draw logic and the wrapping HTML element is the complicated part here.

Current behaviour is that yamcs-web fits the available space, and then the displays crops to those dimensions.

fqqb, 2021-09-04:
Scrollbars, and zoom in/out added on master.

closed 2021-09-04 by fqqb