yamcs/yamcs #505: Adding a command to a stack does not allow assignment of inherited arguments

merose, 2021-02-03:
To reproduce:

- Create a base command A that has two arguments, `x` and `y`.
- Create a derived command B that assigns a value to the `y` argument, but not `x`.
- Create a new command stack.
- Click "ADD COMMAND" within the stack and select command B.

Expected result: You can select a value for argument `x` in the popup dialog.
Actual result: There is only the "Comment" text area, no form field for specifying the value of the `x` argument.

In our case, the `x` argument is an enumerated type, which should present a select list to the user, but no select element is shown. Arguments defined in derived commands do appear, however.

Sending commands using "Commands > Send a command" correctly show form fields for inherited arguments.

merose, 2021-02-03:
Here is the screen when "Send a command" is used, showing a dropdown for specifying the `apid` argument in the "Noop" command:

[image]

Here is the screen when trying to add the "Noop" command to a command stack, showing no dropdown for specifying the apid:

[image]

merose, 2021-02-03:
Here is the XTCE snippet showing the definition of `CommandHeader`, the base command in the `Ccsds` SpaceSystem (some lines omitted):

      <MetaCommand name="CommandHeader" abstract="true">
        <ArgumentList>
          <Argument name="apid" argumentTypeRef="Apid"/>
          <Argument name="opCode" argumentTypeRef="OpCode"/>
        </ArgumentList>
        <CommandContainer name="Header">
          <EntryList>
            ....
          </EntryList>
        </CommandContainer>
      </MetaCommand>

Here is the definition of the derived `Noop` command:

      <MetaCommand name="Noop" shortDescription="No Op command">
        <BaseMetaCommand metaCommandRef="/Ccsds/CommandHeader">
          <ArgumentAssignmentList>
            <ArgumentAssignment argumentName="opCode" argumentValue="APP_NOOP"/>
          </ArgumentAssignmentList>
        </BaseMetaCommand>
        <CommandContainer name="Noop">
          <EntryList>
          </EntryList>
        </CommandContainer>
      </MetaCommand>

fqqb, 2021-02-04:
I'll attempt to reproduce.
It's weird because the underlying form component is identical between both pages (taking as input the same mdb description).

In case you see an error in browser development tools, that could also be useful to know.

xpromache, 2021-02-04:
I don't know if it is related but shouldn't your CommandContainer also specify the "Header" as its BaseContainer?

XTCE is a bit weird in this area.

> Here is the XTCE snippet showing the definition of `CommandHeader`, the base command in the `Ccsds` SpaceSystem (some lines omitted):
>
> ```
>   <MetaCommand name="CommandHeader" abstract="true"> [...]

fqqb, 2021-02-10:
On master. Reason was the MDB object used when coming from a list selectino did not include the parent details.

closed 2021-02-10 by fqqb