yamcs/yamcs #506: CFDP Incoming Transaction Being Cancelled unexpectedly

melekajacob, 2021-02-05:
Hello, I'm having this problem where in the middle of an incoming CFDP transaction, all the incoming packets now are "UNEXPECTED_CFDP_PACKET". See the image below. This happens at different times during transfers, sometimes near the beginning, sometimes near the end.
[image]

However, wireshark shows that the transaction numbers are the same in the valid vs. unexpected packets. 
Good Packet (see its the same packet that was accepted by Yamcs):
[image]
Unexpected Packet (This is the very next packet that wasn't expected):
[image]

I am on Ubuntu 18.04.05 LTS, and this is my CFDP configuration
[image]

From what I'm seeing in the code, I believe the cause is the pending transfer being dropped, but I could be wrong
[image]
.

xpromache, 2021-02-05:
Hello, I think you are running an old version of Yamcs, can you update to the latest (5.3.5)? I fixed a lot of CFDP issues recently.

melekajacob, 2021-02-08:
I was using the 5.2.1 release, I tried resetting to 5.3.5 tag, and was experiencing the same issue

xpromache, 2021-02-08:
Can you please post a log?
The log message from 5.3.5 should be a bit more detailed than 5.2.1 (that's
how I saw you were using an older version).

melekajacob, 2021-02-08:
The logs seem similar, this is the commit I'm at (5.3.5 tag on master branch)
[image]

And these are the logs I'm seeing
[image]

xpromache, 2021-02-08:
That is weird because according to this:
https://github.com/yamcs/yamcs/blob/yamcs-5.3.5/yamcs-core/src/main/java/org/yamcs/cfdp/CfdpService.java#L392

you should have also the header printed on that "Unexpected CFDP packet"
line.

If you go in the admin area from yamcs-web (http://localhost:8090/admin)
what do you see as yamcs version?

melekajacob, 2021-02-08:
Ahh sorry our server was using the older yamcs-core version as a dependency, changing to 5.3.5 fixed this issue. Thank you so much for the help

closed 2021-02-09 by melekajacob