yamcs/yamcs #512: parameter samples response does not contain limit information

nikhilmandlik, 2021-02-10:
parameter samples: https://docs.yamcs.org/yamcs-http-api/parameter-archive/get-parameter-samples/
subscribe parameters: https://docs.yamcs.org/yamcs-http-api/processing/subscribe-parameters/

parameter samples response only returns 
```
avg: 26.799999237060547
max: 26.799999237060547
min: 26.799999237060547
n: 1
time: "2021-02-10T19:04:09.858Z"
```
its missing limit data such as 
```
monitoringResult: MonitoringResult;
rangeCondition: RangeCondition;
alarmRange: AlarmRange[];

etc.
```

This information is important to show limit violations for historical data. (without subscribe parameters or extra calculations on each sample data)

xpromache, 2021-02-10:
The yamcs-web uses the MDB information to draw horizontal limit lines on the plotting page:
https://docs.yamcs.org/yamcs-http-api/mdb/get-parameter/

But I agree, it is not the same because the MDB definition might have changed or the parameter could also have multiple limit sets in the MDB (what XTCE calls contextual alarms).
Still I don't see how the monitoring info could be combined in the same result with the samples (because one sample covers a time interval where monitoring info can change), I would rather see a different call for that - one which can provide a summary of the monitoring info over the [start, stop] period.

merose, 2021-02-10:
Here is the use case. Right now it is specific to OpenMCT, but other clients might also desire this behavior.

To display a plot where the number of data points is larger than the horizontal pixels in the plot window, OpenMCT uses the parameter samples API. The number of samples desired is equal to the number of horizontal pixels available in the plot window.

The horizontal pixel positions each form a time window. At each horizontal pixel position, two data points are plotted, one for the max and one for the min for that interval.

If the user hovers over the min or max plotted point, OpenMCT displays details for that value, including the time. Right now, that time is the midpoint of the sample interval. We would like the time at the min value and the time at the max value also returned by the parameter samples API so that we can display the actual time at which the value was a the minimum or maximum value during that interval. (I realize it could be equal to the min or max at multiple times during the interval. Ideally the time would be the most recent time. We would accept any time at which the value was equal to that min or max, however. We just don't want to mislead the user about the time.)

xpromache, 2021-02-10:
That's right, we discussed having the min/max times as part of the samples,
shouldn't be difficult to do - I'll see to it soon.

But the original question was about the monitoring information such as
limits.
I don't see how that could cleanly be made part of the samples, I rather
see it as a separate call but the display can of course combine the plot
with the limits information.

merose, 2021-02-10:
Ooh, sorry, misunderstood! Inserted myself where I didn't belong.

akhenry, 2021-02-12:
@xpromache I agree, it's a tricky problem. The change that @merose is talking about actually potentially presents a solution though. Right now as I understand it the samples endpoint returns generated timestamps and values, rather than real values. We would like real values (ie. for this time bucket, what were the real minimum and maximum values for this parameter?). Since we would be dealing with real values, would that make it easier to return monitoring results as well?

Without limit evaluation results in sample values we might consider a hybrid approach where limits are monitored by YAMCS for all of our telemetry display types except for plots which would need to duplicate limit evaluation on the client side to work with sample values. In this case we would probably just do it all client side for consistency, but it would be a shame given that YAMCS is providing this information for us most of the time.

xpromache, 2021-02-13:
Ok, that is an idea to have the minimum and maximum as real parameters, not only the engValue. 

Note however that it is not guaranteed that the min and/or max values are also the ones out of limits.

If dynamic (contextual) limits are used, one could have an OOL for a parameter in the middle while the min and max would be within limits. Or, and this may be more common example, there will be no limits defined for some periods of time. 

If dynamic limits are used, you cannot evaluate the out of limits just looking at that parameter. You would need also the parameters which determine the set of limits to be used. If you want to do that on the client I think it will complicate your logic unnecessary.  So we better give all the necessary info from the server.

Do OpenMCT plots work with dynamic limits?

About archive retrieval:

the parameter archive stores the information split in four records (each covering a time interval) for each parameter:
- timestamp
- raw value
- eng value
- parameter status (this includes limits and expiration)

This is done because each of those records can be compressed much better separately than if all info was stored together (parameter status is mostly constant, timestamp monotonically increasing and eng/raw value may also be pretty constant).

Up to recently we were only retrieving the timestamps and engValue to build the samples. A change was made recently to include the parameter status as part of retrieval in order to retrieve the expiration time required to split the samples around gaps (the yamcs-web plots now show interruptions in case of gaps in data).

If we want to reconstitute the original parameter value, we need to retrieve the raw values as well.
But yes, it's doable, no problem, we just need to agree on what to do exactly.

I feel like have we have to go back to my retrieving profile option  - I had at some point made a proposal to have different retrieving profiles which configure which fields are part of the query result. I.e. we would have one profile for yamcs-web, another one for OpenMCT and and another one for Grafana.

@fqqb  didn't like very much my proposal so it was dropped. But still I think if we get all the possible information for all applications even if each is using only parts of it, it will blow up the result.

Another option is having fields turned individually on and off but that will complicate the code on the server with a lot of "if this then add that" statements, whereas my profile option would require a class implemented for each profile.

to be continued...

akhenry, 2021-02-16:
Thanks for considering this request! We appreciate it.

> Note however that it is not guaranteed that the min and/or max values are also the ones out of limits.

Interesting. I'd like to understand a little more about why this is? In any case, I _think_ this is ok, but I'll need to check. By their nature plots hide sub-pixel details, and indicating that a limit violation has occurred, but plotting a different value (the min or max value) would just be misleading, so it would be better not to show the limit violation against that plotted point. I think we would want to show the limit violation in some other way.

> If dynamic limits are used, you cannot evaluate the out of limits just looking at that parameter. You would need also the parameters which determine the set of limits to be used. If you want to do [...]

I agree 100%, we don't want to reconstitute this logic in the client if YAMCS is doing it already.

> Do OpenMCT plots work with dynamic limits?

In principle, yes. Our views invoke a limit function on each parameter value returned from a historical query or received in real time. The limit function returns the result of a limit evaluation that can either be based on a fixed value, or on some function that changes over time. In the case of YAMCS our limit function just reads off the limit status information from the parameter values themselves and returns appropriate limit information, which is nice and easy. Except for samples.

xpromache, 2021-02-16:
> > Note however that it is not guaranteed that the min and/or max values are also the ones out of limits.
> 
> Interesting. I'd like to understand a little more about why this is? 

Well it's because retrieving data from  a large period of time means that each  (min,max, average) sample is aggregating a time interval with many data points possibly each with its own limit. But I understand, you want to get the min and max with their limits and ignore the limits for the other values. Pretty sure this is ok for 99.99% of the cases because the min or the max will be the one out of limits if any.
Another way to do it is to show an indication that an out of limit took place without actually showing the limit - but probably this is an overkill - a bit difficult to do especially if multiple variables are on the same plot.

So looks like we settle on sending full values for the min/max in each sample.

BTW, I looked a bit at your demo https://openmct-demo.herokuapp.com/ and couldn't find any plots showing limits. Are there any?

akhenry, 2021-02-16:
> each (min,max, average) sample is aggregating a time interval with many data points possibly each with its own limit.

Ah, I see, thanks. The limit definition might change over time.

> Pretty sure this is ok for 99.99% of the cases because the min or the max will be the one out of limits if any.
Right. In most cases it's probably not going to be an issue, but users could potentially be looking at weeks worth of data in a plot that is in a dense display, so it's not impossible for this situation to arise if we do adjust limits once we're on the moon (which we almost certainly will I suspect). Let's say one pixel is one hour of data and we raise a limit during one of the hours being plotted, we might hit this issue. Thanks for raising it!

> BTW, I looked a bit at your demo https://openmct-demo.herokuapp.com/ and couldn't find any plots showing limits. Are there any?

Right, that demo does not include limits. If you pull down our main repo and run it locally (requires Node.JS), the "Sine Wave Generator" telemetry point has limits defined at +-0.5 and +-0.9. These limits manifest in all of our telemetry visualizations (plots, tables, latest value tables, alpha-numerics).

xpromache, 2021-02-24:
It seems we cannot reach a common agreement (at SpaceApps) on how to change the standard samples API.

Therefore I propose we make a samples REST endpoint specifically for OpenMCT; it may become part of a future OpenMCT plugin if other "custom" functions are identified.

Or it might be merged back to the main API if we decide it is useful after all.