yamcs/yamcs #514: Exception when trying to compute aggregate member value in MathAlgorithm

merose, 2021-02-15:
Background: I'm trying to construct aggregate parameters based on calculations of parameters from other SpaceSystems, but I'm having difficulties both referencing aggregate parameters from MathAlgorithm or CustomAlgorithm definitions and also storing into output parameters that are members of an aggregate. This report is based on simplifying one of the cases, that of trying to store into an aggregate member.

To reproduce:

1. Get the Yamcs quickstart project.
2. Edit src/main/yamcs/etc/yamcs.yaml, adding a definition of serverId: `serverId: localhost`
3. Modify src/main/yamcs/mdb/xtce.xml by adding a subsidiary SpaceSystem, below, which adds an aggregate parameter with a member that should be updated in a MathAlgorithm.
4. Run Yamcs using `mvn yamcs:run`

Expected result: Aggregate member is updated by the MathAlgorithm in the subsidiary SpaceSystem
Actual result: Exception when starting Yamcs (though Yamcs continues to run)

The subsidiary SpaceSystem, to be inserted into xtce.xml just before the closing `</SpaceSystem>`:

    <SpaceSystem name="Test">
        <TelemetryMetaData>
            <ParameterTypeSet>
                <IntegerParameterType name="Integer" sizeInBits="32" />
                <AggregateParameterType name="Aggregate">
                    <MemberList>
                        <Member typeRef="Integer" name="count" />
                    </MemberList>
                </AggregateParameterType>
            </ParameterTypeSet>
            <ParameterSet>
                <Parameter parameterTypeRef="Aggregate" name="aggregate" />
            </ParameterSet>
            <AlgorithmSet>
                <MathAlgorithm name="DoubleThreadCount">
                    <MathOperation outputParameterRef="aggregate/count">
                        <ParameterInstanceRefOperand parameterRef="/yamcs/localhost/jvmThreadCount"/>
                        <ValueOperand>2</ValueOperand>
                        <Operator>*</Operator>
                        <TriggerSet triggerRate="0">
                            <OnParameterUpdateTrigger parameterRef="/yamcs/localhost/jvmThreadCount"/>
                        </TriggerSet>
                    </MathOperation>
                </MathAlgorithm>
            </AlgorithmSet>
        </TelemetryMetaData>
    </SpaceSystem>

The exception is:

    11:28:17.627 myproject [11] InternalStream [sys_param] Exception received when emitting tuple to subscriber StreamParameterProvider [RUNNING]: java.lang.IllegalStateException: Unknown parameter type 'org.yamcs.xtce.AggregateParameterType@42b3b079'
	org.yamcs.xtceproc.ParameterTypeUtils.getEngValue(ParameterTypeUtils.java:263)
	org.yamcs.algorithms.MathAlgorithmExecutor.runAlgorithm(MathAlgorithmExecutor.java:48)
	org.yamcs.algorithms.AlgorithmManager.updateParameters(AlgorithmManager.java:410)
	org.yamcs.algorithms.AlgorithmManager.updateParameters(AlgorithmManager.java:388)
	org.yamcs.parameter.ParameterRequestManager.update(ParameterRequestManager.java:440)
	org.yamcs.tctm.StreamParameterProvider.onTuple(StreamParameterProvider.java:139)
	org.yamcs.yarch.Stream.emitTuple(Stream.java:56)
	org.yamcs.parameter.SystemParametersCollector.run(SystemParametersCollector.java:218)
	java.util.concurrent.Executors$RunnableAdapter.call(Executors.java:511)
	java.util.concurrent.FutureTask.runAndReset(FutureTask.java:308)
	java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.access$301(ScheduledThreadPoolExecutor.java:180)
	java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.run(ScheduledThreadPoolExecutor.java:294)
	java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1149)
	java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:624)
	java.lang.Thread.run(Thread.java:748)

If the reference to the aggregate member as output to the MathOperation is changed to use a period instead of a slash:

    <MathOperation outputParameterRef="aggregate.count">

then the exception reported is different:

    11:33:58.932 _global [1] YamcsServer Instance myproject failed: org.yamcs.xtce.DatabaseLoadException: Cannot resolve reference SpaceSystem: Test name: aggregate.count type: PARAMETER
	org.yamcs.xtceproc.XtceDbFactory.resolveReferences(XtceDbFactory.java:233)
	org.yamcs.xtceproc.XtceDbFactory.resolveReferences(XtceDbFactory.java:248)
	org.yamcs.xtceproc.XtceDbFactory.resolveReferences(XtceDbFactory.java:248)
	org.yamcs.xtceproc.XtceDbFactory.createInstance(XtceDbFactory.java:151)
	org.yamcs.xtceproc.XtceDbFactory.getInstance(XtceDbFactory.java:428)
	org.yamcs.YamcsServerInstance.doInit(YamcsServerInstance.java:135)
	org.yamcs.YamcsInstanceService.initAsync(YamcsInstanceService.java:62)
	org.yamcs.YamcsServerInstance.init(YamcsServerInstance.java:120)
	org.yamcs.YamcsServer.addInstance(YamcsServer.java:576)
	org.yamcs.YamcsServer.addGlobalServicesAndInstances(YamcsServer.java:1271)
	org.yamcs.YamcsServer.prepareStart(YamcsServer.java:1107)
	org.yamcs.YamcsServer.main(YamcsServer.java:909)
    11:33:58.932 _global [1] YamcsServer Failure while attempting to validate configuration: org.yamcs.xtce.DatabaseLoadException: Cannot resolve reference SpaceSystem: Test name: aggregate.count type: PARAMETER
	org.yamcs.xtceproc.XtceDbFactory.resolveReferences(XtceDbFactory.java:233)
	org.yamcs.xtceproc.XtceDbFactory.resolveReferences(XtceDbFactory.java:248)
	org.yamcs.xtceproc.XtceDbFactory.resolveReferences(XtceDbFactory.java:248)
	org.yamcs.xtceproc.XtceDbFactory.createInstance(XtceDbFactory.java:151)
	org.yamcs.xtceproc.XtceDbFactory.getInstance(XtceDbFactory.java:428)
	org.yamcs.YamcsServerInstance.doInit(YamcsServerInstance.java:135)
	org.yamcs.YamcsInstanceService.initAsync(YamcsInstanceService.java:62)
	org.yamcs.YamcsServerInstance.init(YamcsServerInstance.java:120)
	org.yamcs.YamcsServer.addInstance(YamcsServer.java:576)
	org.yamcs.YamcsServer.addGlobalServicesAndInstances(YamcsServer.java:1271)
	org.yamcs.YamcsServer.prepareStart(YamcsServer.java:1107)
	org.yamcs.YamcsServer.main(YamcsServer.java:909)

In this case Yamcs fails to start.

xpromache, 2021-02-15:
We consider that aggregate values should be provided in "full" and I don't see how this could be possible via MathAlgorithm which only calculates one single double value. In your example, the aggregate has only one member but in general if there are more how is this supposed to work?
An analogy (in my mind) is if you had a 32 bit parameter and an algorithm would set only 3 bits.

I guess if the parameter had a default value, we could allow updating just a member.

In any case it shouldn't throw a stack trace like that so I will fix that.

merose, 2021-02-15:
In our case we're composing an aggregate based on rover pose information and image information, so each algorithm can only fill in part of the aggregate. But I understand your point. That's a limitation we could live with, though it would mean we would have to write the algorithm code in Java or a scripting language and could not use MathAlgorithm even if it would accomplish part of what we want to do.

merose, 2021-02-15:
I'm running into various issues with algorithms and so am filing reports based on what I find, even if it is not the end result we are trying to achieve. I understand if some reports may be rejected as "will not fix" because it is not intended to work that way.