yamcs/yamcs #515: Cannot refer to aggregate member in other SpaceSystem from MathAlgorithm

merose, 2021-02-15:
To reproduce:

1. Get the Yamcs quickstart project.
2. Modify pom.xml to refer to the latest Yamcs version (5.4.0, right now).
3. Add two subsidiary SpaceSystems to src/main/yamcs/mdb/xtce.xml, as shown below.
4. Start Yamcs using `mvn yamcs:run`

Expected result: Yamcs starts and MathAlgorithm runs once per second.
Actual result: Yamcs fails to start with the exception below.

The subsidiary SpaceSystems to be added just before the closing `</SpaceSystem`:

    <SpaceSystem name="Test1">
        <TelemetryMetaData>
            <ParameterTypeSet>
                <IntegerParameterType name="Integer1" sizeInBits="32" />
                <AggregateParameterType name="Aggregate1">
                    <MemberList>
                        <Member typeRef="Integer1" name="i1" initialValue="7" />
                    </MemberList>
                </AggregateParameterType>
            </ParameterTypeSet>
            <ParameterSet>
                <Parameter parameterTypeRef="Aggregate1" name="aggregate1" />
            </ParameterSet>
        </TelemetryMetaData>
    </SpaceSystem>
    
    <SpaceSystem name="Test2">
        <TelemetryMetaData>
            <ParameterTypeSet>
                <IntegerParameterType name="Integer2" sizeInBits="32" />
            </ParameterTypeSet>
            <ParameterSet>
                <Parameter parameterTypeRef="Integer2" name="i2" />
            </ParameterSet>
            <AlgorithmSet>
                <MathAlgorithm name="Double">
                    <MathOperation outputParameterRef="i2">
                        <ParameterInstanceRefOperand parameterRef="/myproject/Test1/aggregate1/i1"/>
                        <ValueOperand>2</ValueOperand>
                        <Operator>*</Operator>
                        <TriggerSet triggerRate="0">
                            <OnPeriodicRateTrigger fireRateInSeconds="1"/>
                        </TriggerSet>
                    </MathOperation>
                </MathAlgorithm>
            </AlgorithmSet>
        </TelemetryMetaData>
    </SpaceSystem>

The exception:

    11:55:23.998 _global [1] XtceDbFactory Found eference to an aggregate member using path separator '/': /myproject/Test1/aggregate1/i1. Please use dot '.' separator instead.
    11:55:23.999 _global [1] YamcsServer Instance myproject failed: java.lang.ClassCastException: org.yamcs.xtce.util.UnresolvedNameReference cannot be cast to org.yamcs.xtce.util.UnresolvedParameterReference
	org.yamcs.xtceproc.XtceDbFactory.resolveReferences(XtceDbFactory.java:240)
	org.yamcs.xtceproc.XtceDbFactory.resolveReferences(XtceDbFactory.java:249)
	org.yamcs.xtceproc.XtceDbFactory.resolveReferences(XtceDbFactory.java:249)
	org.yamcs.xtceproc.XtceDbFactory.createInstance(XtceDbFactory.java:152)
	org.yamcs.xtceproc.XtceDbFactory.getInstance(XtceDbFactory.java:435)
	org.yamcs.YamcsServerInstance.doInit(YamcsServerInstance.java:135)
	org.yamcs.YamcsInstanceService.initAsync(YamcsInstanceService.java:62)
	org.yamcs.YamcsServerInstance.init(YamcsServerInstance.java:120)
	org.yamcs.YamcsServer.addInstance(YamcsServer.java:583)
	org.yamcs.YamcsServer.addGlobalServicesAndInstances(YamcsServer.java:1258)
	org.yamcs.YamcsServer.prepareStart(YamcsServer.java:1089)
	org.yamcs.YamcsServer.main(YamcsServer.java:891)
    11:55:24.000 _global [1] YamcsServer Failure while attempting to validate configuration: java.lang.ClassCastException: org.yamcs.xtce.util.UnresolvedNameReference cannot be cast to org.yamcs.xtce.util.UnresolvedParameterReference
	org.yamcs.xtceproc.XtceDbFactory.resolveReferences(XtceDbFactory.java:240)
	org.yamcs.xtceproc.XtceDbFactory.resolveReferences(XtceDbFactory.java:249)
	org.yamcs.xtceproc.XtceDbFactory.resolveReferences(XtceDbFactory.java:249)
	org.yamcs.xtceproc.XtceDbFactory.createInstance(XtceDbFactory.java:152)
	org.yamcs.xtceproc.XtceDbFactory.getInstance(XtceDbFactory.java:435)
	org.yamcs.YamcsServerInstance.doInit(YamcsServerInstance.java:135)
	org.yamcs.YamcsInstanceService.initAsync(YamcsInstanceService.java:62)
	org.yamcs.YamcsServerInstance.init(YamcsServerInstance.java:120)
	org.yamcs.YamcsServer.addInstance(YamcsServer.java:583)
	org.yamcs.YamcsServer.addGlobalServicesAndInstances(YamcsServer.java:1258)
	org.yamcs.YamcsServer.prepareStart(YamcsServer.java:1089)
	org.yamcs.YamcsServer.main(YamcsServer.java:891)

merose, 2021-02-15:
Changing the parameter reference to use a "." instead of "/" (which makes it invalid XTCE, as you've noted in the code), fails with the same exception.

    <ParameterInstanceRefOperand parameterRef="/myproject/Test1/aggregate1.i1"/>

merose, 2021-02-15:
Looking through XtceStaxReader.java, it appears that the 7 instantiations of UnresolvedNameReference could be replaced with UnresolvedParameterReference:

    $ egrep -n 'new UnresolvedNameReference.*Type.\<PARAMETER\>' yamcs-xtce/src/main/java/org/yamcs/xtce/xml/XtceStaxReader.java 
    1704:            NameReference nr = new UnresolvedNameReference(refName, Type.PARAMETER).addResolvedAction(nd -> {
    1738:        NameReference nr = new UnresolvedNameReference(refName, Type.PARAMETER).addResolvedAction(nd -> {
    2661:        NameReference nr = new UnresolvedNameReference(refName, Type.PARAMETER).addResolvedAction(nd -> {
    2752:        NameReference nr = new UnresolvedNameReference(refName, Type.PARAMETER).addResolvedAction(nd -> {
    2879:        NameReference nr = new UnresolvedNameReference(paramRef, Type.PARAMETER).addResolvedAction(nd -> {
    3946:        NameReference nr = new UnresolvedNameReference(paramRef, Type.PARAMETER).addResolvedAction(nd -> {
    4015:        NameReference nr = new UnresolvedNameReference(paramRef, Type.PARAMETER).addResolvedAction(nd -> {

xpromache, 2021-02-24:
Thanks for the report. I think this is fixed in master (I wrongly made the reference to #514).

xpromache, 2021-02-27:
Pushed to master some changes to allow aggregate values to be used in full (not only members) in Java algorithms. 

See an example in [ref-xtce.xml](https://github.com/yamcs/yamcs/blob/master/yamcs-core/src/test/resources/xtce/ref-xtce.xml#L81) and [RefXtceAlgorithmTest](https://github.com/yamcs/yamcs/blob/master/yamcs-core/src/test/java/org/yamcs/algorithms/RefXtceAlgorithmTest.java#L160)

Not yet working in Javascript or Python.

xpromache, 2021-04-20:
this should be solved on master, please reopen if not.

closed 2021-04-20 by xpromache