yamcs/yamcs #518: Cannot retrieve initial value of parameter in MathAlgorithm

merose, 2021-02-23:
To reproduce:

- In the quickstart project, add the following SpaceSystem to the end of xtce.xml. The MathAlgorithm attempts to copy the parameter value from `value` to `valueCopy`, but it will fail.

~~~
  <SpaceSystem name="Test">
    <TelemetryMetaData>
      <ParameterTypeSet>
        <FloatParameterType name="Number" />
      </ParameterTypeSet>
      <ParameterSet>
        <Parameter parameterTypeRef="Number" name="value" initialValue="3.14" />
        <Parameter parameterTypeRef="Number" name="valueCopy" />
      </ParameterSet>
      <AlgorithmSet>
        <MathAlgorithm name="CopyValue">
          <MathOperation outputParameterRef="valueCopy">
            <ParameterInstanceRefOperand parameterRef="value" useCalibratedValue="true"/>
            <TriggerSet triggerRate="0">
              <OnPeriodicRateTrigger fireRateInSeconds="1"/>
            </TriggerSet>
          </MathOperation>
        </MathAlgorithm>
      </AlgorithmSet>
    </TelemetryMetaData>
  </SpaceSystem>
~~~

- Start Yamcs and look at the telemetry value `/myproject/Test/value` and at the value `/myproject/Test/valueCopy`.

Expected result: `value` has the value 3.14 and `valueCopy` also has the value 3.14. `valueCopy` is updated once per second with the same value.
Actual result: `value` is 3.14, but `valueCopy` is always zero. It _is_ updated once per second, but always with the value zero.

-----
A clue: If the original parameter is set by an algorithm rather than by `initialValue`, it works.

- Change the `Test` SpaceSystem to be the following. In this case a second MathAlgorithm sets the value of the `value` parameter to 3.14 instead of using `initialValue`. When the other algorithm copies the value, it works.

~~~
  <SpaceSystem name="Test">
    <TelemetryMetaData>
      <ParameterTypeSet>
        <FloatParameterType name="Number" />
      </ParameterTypeSet>
      <ParameterSet>
        <Parameter parameterTypeRef="Number" name="value" />
        <Parameter parameterTypeRef="Number" name="valueCopy" />
      </ParameterSet>
      <AlgorithmSet>
        <MathAlgorithm name="SetValue">
          <MathOperation outputParameterRef="value">
            <ValueOperand>3.14</ValueOperand>
            <TriggerSet>
              <OnPeriodicRateTrigger fireRateInSeconds="10"/>
            </TriggerSet>
          </MathOperation>
        </MathAlgorithm>
        <MathAlgorithm name="CopyValue">
          <MathOperation outputParameterRef="valueCopy">
            <ParameterInstanceRefOperand parameterRef="value" useCalibratedValue="true"/>
            <TriggerSet triggerRate="0">
              <OnPeriodicRateTrigger fireRateInSeconds="1"/>
            </TriggerSet>
          </MathOperation>
        </MathAlgorithm>
      </AlgorithmSet>
    </TelemetryMetaData>
  </SpaceSystem>
~~~

- Start Yamcs and view the parameters `/myproject/Test/value` and `/myproject/Test/valueCopy`.

Expected and actual result: Both `value` and `valueCopy` have the value 3.14.

merose, 2021-02-23:
A curious additional observation: when running the 2nd XTCE from above, the historical values of the `value` parameter do not get updated. That is, the value is set only once, not once every 10 seconds as configured. (This is using Yamcs 5.4.0.)

xpromache, 2021-02-24:
Thanks for the report. Should be fixed on master.

xpromache, 2021-03-10:
Please verify that this is working as expected in latest version 5.4.2.
Closing for now.

closed 2021-03-10 by xpromache