yamcs/yamcs #522: MathOperator.OVER implementation appears wrong

merose, 2021-02-26:
In MathOperator.java, the documentation for MathOperator.OVER is this, which matches my understanding of similar languages such as PostScript (line 34):

    OVER("over", 1), //Duplicate top item on the stack (x1 x2 -- x1 x2 x1)

However, the implementation in MathOperationCalibratorFactory.java appears wrong. It appears that the stack effects will be: x1 x2 -- x2 x1 x2. Here is the code (lines 77-82):

                } else if (mo == MathOperator.OVER) {
                    String x1 = stack.pop();
                    String x2 = stack.pop();
                    stack.push(x1);
                    stack.push(x2);
                    stack.push(x1);

I believe it would be correct, and clearer, to have the code look like this, which matches the documentation in MathOperator.OVER:

                } else if (mo == MathOperator.OVER) {
                    // x1 x2 -- x1 x2 x1
                    String x2 = stack.pop();
                    String x1 = stack.pop();
                    stack.push(x1);
                    stack.push(x2);
                    stack.push(x1);

-----

Similarly, the code for MathOperator.SWAP, though correct, does not match the documentation of MathOperator.SWAP. The documentation looks like this:

    SWAP("swap", 2), //swap the top two stack items (x1 x2 -- x2 x1)

But the code reverses the names of the two temporary variables:

                if (mo == MathOperator.SWAP) {
                    String x1 = stack.pop();
                    String x2 = stack.pop();
                    stack.push(x1);
                    stack.push(x2);

It would be more clearly correct to match the documentation like this:

                if (mo == MathOperator.SWAP) {
                    // x1 x2 -- x2 x1
                    String x2 = stack.pop();
                    String x1 = stack.pop();
                    stack.push(x2);
                    stack.push(x1);

xpromache, 2021-02-26:
The comments in the code are coming straight off the XTCE 1.2 XSD:
https://github.com/yamcs/yamcs/blob/master/yamcs-xtce/SpaceSystemV1.2.xsd

I think they intend them that the top of the stack is at the left. For example in:

`(x1 x2 -- x1 x2 x1)
`
x1 is on the top of the stack. 

I base myself on the description of the subtraction operation on line 4689:
`subtraction (x1 x2 -- x1-x2)
`

Moreover your implementation of OVER seems to duplicate not the top of the stack which is the first element popped (x2 in your case) but the second element popped so it does not seem to be doing what the description says.

Am I missing something?

merose, 2021-02-26:
Hmmm... that's not the convention that some other stack-oriented languages use. For example, the Postscript reference always has the top of the stack to the right, and has this description of the "sub" operator (subtraction):

[image]

And your implementation of binary operators that correspond to Java operators is not as you described. Look at the code for a binary operator:

                    } else if (mo.getArity() == 2) {
                        String r = stack.pop();
                        String l = stack.pop();
                        stack.push(getJavaExpression(mo, l, r));

and the resulting expression:

    return "(" + l + " " + op.xtceName() + " " + r + ")";

This implementation generates "2nd from top - top of stack", which is what I would expect (based on how stack operations work in Postscript, Forth, and HP calculators), and corresponds to the interpretation that the top of the stack is rightmost in the stack operation description: `x1 x2 -- x1-x2`

The Forth language also has the `over` operator. The Forth documentation always shows the top of the stack at the right of a sequence. The definition of over is:

    over  ( a b -- a b a )

In other words, `over` takes the 2nd-from-top and pushes that on top of the stack. I believe that is what is intended in XTCE as well based on the operation definition: `x1 x2 -- x1 x2 x1`

xpromache, 2021-02-26:
Ok so indee I did miss something. 
Thanks for the explanations, I will change it as per your suggestions.

commit 5487d12c (2021-02-27): fixed the MathOperator.OVER implementation

Also improved the docs
Fixes #522
closed 2021-02-27 by xpromache