yamcs/yamcs #524: Does Yamcs support segmentation of large commands across multiple TC Frames?

AdamElliott-AB, 2021-03-02:
According to https://github.com/yamcs/yamcs/issues/398 Yamcs supports packets segmented across frames for telemetry. Does Yamcs support segmentation for commanding as well using the optional Segment Header of the "TC Space Data Link Protocol"

xpromache, 2021-03-02:
No, that is not supported yet;
currently one TC packet = one TC frame.

xpromache, 2021-03-30:
> No, that is not supported yet;
> currently one TC packet = one TC frame.

I fixed a bug related to the maximum frame size (sending a command that would result exactly in that maximum size failed due to a > instead of a >= condition) and noticed that what I wrote above is not exactly true: multiple small commands can be fit into one frame but one big command will not be split across multiple frames.