yamcs/yamcs #532: Parameter archive lags by approx. 25 min for some parameters

merose, 2021-03-09:
For some parameters we're seeing a gap in archived values available, starting 15 minutes prior to current wall-clock time and extending back by 25 additional minutes. (So between now minus 40min and now minus 15min there are no values available.) However, this looks like a moving gap: older values are filled in, but newer values drop out. It may be related to the difference between the in-memory cache and the permanent archive.

This is probably some misconfiguration on my part. Could this be related to the parameter cache settings, or parameter archive settings, or the parameter archive filler settings, or something else?

merose, 2021-03-10:
After further investigation, I think it may be a retrieval issue rather than a storage issue. Still trying to characterize further. I realize this is very vague, but I'm definitely seeing a problem, both in the plot view in the Yamcs UI (which uses Get Parameter Samples) and also in the HTTP API using List Parameter History. It is strange that it's appearing on a parameter with a fairly low update rate, approx. .3 Hz on average, when I have other parameters that have a 10Hz update rate and they are fine.

xpromache, 2021-03-10:
I was planning for a while to implement a few REST calls to provide more
information about the parameter archive:
 - which parameters are there (param qualified name, rawType, engType) -
note that it might be different from the current MDB
 - which records are available for each parameter: (start, stop, sample
count)

This looks like a good time... so if you can wait a couple of days and try
with a non-released version of Yamcs, it might help debug the problem.

merose, 2021-03-10:
Sure. I can replicate this pretty easily. But I may not get to it until next week. We have an internal release due early next week, and I have a lot of test setup to finish. BTW, we will be testing with the DSN soon. I'll let you know how that goes.

xpromache, 2021-03-10:
So, I made a branch parchive-info where you can do the following:
1. get all the internal ids for the parameter in question:
```
# curl http://localhost:8090/api/archive/simulator/parameterArchive/info/parameters?q=ccsds-seqcount
{
  "parameters": [{
    "id": 71,
    "fqn": "/YSS/ccsds-seqcount",
    "rawType": "UINT32",
    "engType": "UINT32"
  }]
}
```
2. for one id get a summary of all the records in the archive:
```
curl "http://localhost:8090/api/archive/simulator/parameterArchive/info/segments?id=71&start=2021-02-24T17:37:26.360Z"
{
  "segments": [{
    "groupId": 12,
    "start": "2021-02-24T17:04:06.008Z",
    "end": "2021-02-24T17:20:45.761Z",
    "count": 1000
  }, {
    "groupId": 12,
    "start": "2021-02-24T17:20:46.161Z",
    "end": "2021-02-24T17:37:25.761Z",
    "count": 1000
  }, {
    "groupId": 12,
    "start": "2021-02-24T17:37:26.360Z",
    "end": "2021-02-24T17:54:05.761Z",
    "count": 1000
  },
...
```

I'm curios if you get only one id (multiple means that somehow the parameter changes its type) and how many group ids you get.

I wand to implement a few more parameters and then I will commit it to master.

xpromache, 2021-03-11:
In the meanwhile I merged the new methods to the master and added some additional ones. 

There is also some explanation on how the data is stored in the parameter archive and what those parameter id and group id mean:
https://docs.yamcs.org/yamcs-http-api/parameter-archive/get-archived-parameters-info/

xpromache, 2021-04-22:
Hi Marc, did you find more about this problem?
Is there a issue in Yamcs?

merose, 2021-04-22:
I apologize, I have not yet been able to give this more detailed study. I will have time early next week to duplicate the issue again and use your new API calls to see if they give me more info.