yamcs/yamcs #534: Suggestions for packetviewer improvements

merose, 2021-03-10:
I find the packet viewer useful for viewing files of recorded data. However, there are a few changes that would make the packet viewer more useful.

* _Show the correct container qualified name._ Currently if you have a hierarchy of containers in the XTCE, the top
container name in the hierarchy is shown for all packets, instead of the most specific packet name. as well, only the
last component of the qualified name is shown. We have dozens of containers that have similar names that are
differentiated by their fully-qualified name, based on the hierarchy of SpaceSystems. As an example, here is the
packet viewer showing packets in our project. Even though we have over 300 container types, they are all shown as
`TelemetryHeader`.

[image]

* _Show aggregate members on multiple lines._ We have some large aggregates that are difficult to see in the one-line view currently used by the packet viewer. Perhaps have a "+" icon to expand into a multi-line view to show members. As an
example, you can see that the `pose` aggregate parameter cannot be completely shown without scrolling even when
expanding the `EngValue` column to be extremely wide.

[image]

* _Allow opening a packet file while using the MDB from the server._ If you already have a running Yamcs server, it would allow immediate use of the packet viewer if you could get the MDB info from the server, but then open a local file to examine recorded packets. Without this feature, you have to configure the packet viewer by copying your XTCE and editing `packet-viewer.yaml` to match your server YAML. Not a big problem to set up, but it would be convenient if you could point a colleague at the packet viewer download from GitHub and let the configuration happen more-or-less automatically by pulling from the server. Caching that for off-line use would be nice, too.

The last suggestion is for Yamcs, not the packet viewer:

* _Add packet-viewer-like display to Yamcs packet view._ It would be very convenient to be able to view packets in the Yamcs `Telemetry > Packets` view, but when clicking on a packet see not only the raw bytes but also the unpacked parameters. (That would be much appreciated when debugging problems with particular packets.)

fqqb, 2023-07-16:
> Allow searching for packets by container qualified name. The parameter search is great. Searching by packet name like in Yamcs would also be nice.

This is possible by using the expression `packet.name`. A full description of the filter expressions was recently added to the documentation: https://docs.yamcs.org/yamcs-server-manual/programs/packet-viewer/#filter-on-packet-properties

fqqb, 2023-07-28:
> Show the correct container qualified name.

This change has been applied on master. This also means that the use of `packet.name` in the expression filter is now matched against the full name.