yamcs/yamcs #537: WebApp Homepage Missing Info on Latest Build

lorenzo-gomez-windhover, 2021-03-18:
Hi there,

I just installed 2f6f9953bd00755a3324d435cf1544e7dffc84ac and when I start up YAMCS, I get the following:

[image]

As you can see, it only shows information about the CCSDS header(which is the very first row), but not the rest. It does show Parameter information when I go to the `Paramaters ` section of the sidebar, but it does not show the information on Home anymore on this build. If I build any older version of YAMCS(such as ac64eeb295ca20093c4a47e2156da0fc6e323adf) it works. It looks like the latest build might have done this.

Hope the issue makes sense.
Lorenzo

xpromache, 2021-03-18:
Are you using container inheritance or composition? For the difference please read this: 
https://docs.yamcs.org/yamcs-server-manual/processors/tm-processing/

I did change the behavior because the old one was not correct. Now the packets which show on that page are only those inheriting from the container received via data link and passed on the stream (the one that is configured with the rootContainer property of the stream). 

If that container is decomposed (using container composition) in one ore more containers (covering partially the packet) then those will not appear.

The main reason I did that is because the binary packets are stored in the archive with that name, so it would be wrong to use the name of a container that covers only partially the packet to store the whole packet.

lorenzo-gomez-windhover, 2021-03-18:
Unless I misunderstand the XTCE standard,  I'm pretty sure we use inheritance for our containers:

```
              <SequenceContainer name="AK8963_DIAG_TLM_MID">
                    <EntryList>
                        <ParameterRefEntry parameterRef="AK8963_DiagPacket_t"/>
                    </EntryList>
                    <BaseContainer containerRef="cfs-ccsds/cfs-tlm-hdr">
                        <RestrictionCriteria>
                            <ComparisonList>
                                <Comparison parameterRef="cfs-ccsds/ccsds-apid" value="275"/>
                            </ComparisonList>
                        </RestrictionCriteria>
                    </BaseContainer>
                </SequenceContainer>
```

However, as you can see we do use composition with ParameterRefEntry* types that are defined in the namespace/spacesystem. Is this the composition you are referring to?

xpromache, 2021-03-18:
What I mean with composition is to have a ContainerRefEntry in the EntryList.

With your XTCE it appears that there is a bug in the code I changed, I will investigate.

lorenzo-gomez-windhover, 2021-03-18:
Thanks so much for the brief response! Good to know :+1:

xpromache, 2021-03-18:
Can you please verify it is fixed in the last push?

closed 2021-03-18 by xpromache

lorenzo-gomez-windhover, 2021-03-18:
Works like a charm now on 2334e83406d27c9b760de2cb1d0ff48b090dc859. Thanks!

commit 2334e834 (2021-03-18): Fixed (again) the auto-partitioning

Fixed #537 introduced by 8547d7c8bf9bb0e45ba97cd25cf8caa0730097bb - the decision on which container to put the partition flag was taken before the whole tree was built, causing to incorrectly skip some containers.