yamcs/yamcs #538: Cannot select aggregate members when adding a chart comparison

merose, 2021-03-20:
To reproduce:

1. Have an aggregate telemetry parameter "sample" with two numeric members "a" and "b".
2. Feed telemetry for "sample" with values for both a and b.
3. Navigate to Telemetry > Parameters, and navigate to "sample".
4. Click on member "a" and show the Chart view.
5. Click the "+ Comparison" button and search for "sample".

Expected result: You can select sample.a or sample.b.
Actual result: You only see sample in the search list.

6. Select sample, but edit the "Seach Parameter for Comparison" field to end with "sample.b", then click Add.

Expected result: sample.a and sample.b are shown in a comparison chart.
Actual result: No comparison is added to the chart.

merose, 2021-03-20:
There is also an exception shown in the log.

```
14:15:37.793 _global [49] RouteHandler 127: Responding '500 Internal Server Error': java.lang.IllegalStateException: Unnown type AGGREGATE: org.yamcs.http.InternalServerErrorException: java.lang.IllegalStateException: Unnown type AGGREGATE
	org.yamcs.http.RouteHandler.handleException(RouteHandler.java:118)
	org.yamcs.http.RouteHandler.dispatch(RouteHandler.java:87)
	org.yamcs.http.RouteHandler.handle(RouteHandler.java:51)
	org.yamcs.http.Handler.handle(Handler.java:23)
	org.yamcs.http.Handler.channelRead0(Handler.java:29)
	org.yamcs.http.Handler.channelRead0(Handler.java:9)
	io.netty.channel.SimpleChannelInboundHandler.channelRead(SimpleChannelInboundHandler.java:99)
	io.netty.channel.AbstractChannelHandlerContext.invokeChannelRead(AbstractChannelHandlerContext.java:379)
	io.netty.channel.AbstractChannelHandlerContext.invokeChannelRead(AbstractChannelHandlerContext.java:365)
	io.netty.channel.AbstractChannelHandlerContext.fireChannelRead(AbstractChannelHandlerContext.java:357)
	io.netty.handler.codec.MessageToMessageDecoder.channelRead(MessageToMessageDecoder.java:103)
	io.netty.channel.AbstractChannelHandlerContext.invokeChannelRead(AbstractChannelHandlerContext.java:379)
	io.netty.channel.AbstractChannelHandlerContext.invokeChannelRead(AbstractChannelHandlerContext.java:365)
	io.netty.channel.AbstractChannelHandlerContext.fireChannelRead(AbstractChannelHandlerContext.java:357)
	io.netty.handler.codec.MessageToMessageDecoder.channelRead(MessageToMessageDecoder.java:103)
	io.netty.handler.codec.MessageToMessageCodec.channelRead(MessageToMessageCodec.java:111)
	io.netty.channel.AbstractChannelHandlerContext.invokeChannelRead(AbstractChannelHandlerContext.java:379)
	io.netty.channel.AbstractChannelHandlerContext.invokeChannelRead(AbstractChannelHandlerContext.java:365)
	io.netty.channel.AbstractChannelHandlerContext.fireChannelRead(AbstractChannelHandlerContext.java:357)
	org.yamcs.http.HttpRequestHandler.channelRead(HttpRequestHandler.java:158)
	io.netty.channel.AbstractChannelHandlerContext.invokeChannelRead(AbstractChannelHandlerContext.java:379)
	io.netty.channel.AbstractChannelHandlerContext.invokeChannelRead(AbstractChannelHandlerContext.java:365)
	io.netty.channel.AbstractChannelHandlerContext.fireChannelRead(AbstractChannelHandlerContext.java:357)
	io.netty.channel.CombinedChannelDuplexHandler$DelegatingChannelHandlerContext.fireChannelRead(CombinedChannelDuplexHandler.java:436)
	io.netty.handler.codec.ByteToMessageDecoder.fireChannelRead(ByteToMessageDecoder.java:324)
	io.netty.handler.codec.ByteToMessageDecoder.channelRead(ByteToMessageDecoder.java:296)
	io.netty.channel.CombinedChannelDuplexHandler.channelRead(CombinedChannelDuplexHandler.java:251)
	io.netty.channel.AbstractChannelHandlerContext.invokeChannelRead(AbstractChannelHandlerContext.java:379)
	io.netty.channel.AbstractChannelHandlerContext.invokeChannelRead(AbstractChannelHandlerContext.java:365)
	io.netty.channel.AbstractChannelHandlerContext.fireChannelRead(AbstractChannelHandlerContext.java:357)
	io.netty.handler.traffic.AbstractTrafficShapingHandler.channelRead(AbstractTrafficShapingHandler.java:513)
	io.netty.channel.AbstractChannelHandlerContext.invokeChannelRead(AbstractChannelHandlerContext.java:379)
	io.netty.channel.AbstractChannelHandlerContext.invokeChannelRead(AbstractChannelHandlerContext.java:365)
	io.netty.channel.AbstractChannelHandlerContext.fireChannelRead(AbstractChannelHandlerContext.java:357)
	io.netty.channel.DefaultChannelPipeline$HeadContext.channelRead(DefaultChannelPipeline.java:1410)
	io.netty.channel.AbstractChannelHandlerContext.invokeChannelRead(AbstractChannelHandlerContext.java:379)
	io.netty.channel.AbstractChannelHandlerContext.invokeChannelRead(AbstractChannelHandlerContext.java:365)
	io.netty.channel.DefaultChannelPipeline.fireChannelRead(DefaultChannelPipeline.java:919)
	io.netty.channel.nio.AbstractNioByteChannel$NioByteUnsafe.read(AbstractNioByteChannel.java:166)
	io.netty.channel.nio.NioEventLoop.processSelectedKey(NioEventLoop.java:719)
	io.netty.channel.nio.NioEventLoop.processSelectedKeysOptimized(NioEventLoop.java:655)
	io.netty.channel.nio.NioEventLoop.processSelectedKeys(NioEventLoop.java:581)
	io.netty.channel.nio.NioEventLoop.run(NioEventLoop.java:493)
	io.netty.util.concurrent.SingleThreadEventExecutor$4.run(SingleThreadEventExecutor.java:989)
	io.netty.util.internal.ThreadExecutorMap$2.run(ThreadExecutorMap.java:74)
	io.netty.util.concurrent.FastThreadLocalRunnable.run(FastThreadLocalRunnable.java:30)
	java.lang.Thread.run(Thread.java:748)
Caused by: java.lang.IllegalStateException: Unnown type AGGREGATE
	org.yamcs.parameter.ValueArray.newObj(ValueArray.java:181)
	org.yamcs.parameter.ValueArray.<init>(ValueArray.java:25)
	org.yamcs.http.api.SingleParameterRetriever.sendToConsumer(SingleParameterRetriever.java:128)
	org.yamcs.http.api.SingleParameterRetriever.splitAndSend(SingleParameterRetriever.java:121)
	org.yamcs.http.api.SingleParameterRetriever.retrieve(SingleParameterRetriever.java:91)
	org.yamcs.http.api.ParameterArchiveApi.getParameterSamples(ParameterArchiveApi.java:196)
	org.yamcs.http.api.ParameterArchiveApi.getParameterSamples(ParameterArchiveApi.java:62)
	org.yamcs.protobuf.AbstractParameterArchiveApi.callMethod(AbstractParameterArchiveApi.java:161)
	org.yamcs.http.RouteHandler.dispatch(RouteHandler.java:84)
	org.yamcs.http.RouteHandler.handle(RouteHandler.java:51)
	org.yamcs.http.Handler.handle(Handler.java:23)
[... cut ...]
	io.netty.channel.nio.NioEventLoop.processSelectedKeysOptimized(NioEventLoop.java:655)
	io.netty.channel.nio.NioEventLoop.processSelectedKeys(NioEventLoop.java:581)
	io.netty.channel.nio.NioEventLoop.run(NioEventLoop.java:493)
	io.netty.util.concurrent.SingleThreadEventExecutor$4.run(SingleThreadEventExecutor.java:989)
	io.netty.util.internal.ThreadExecutorMap$2.run(ThreadExecutorMap.java:74)
	io.netty.util.concurrent.FastThreadLocalRunnable.run(FastThreadLocalRunnable.java:30)
	java.lang.Thread.run(Thread.java:748)

```

fqqb, 2021-04-20:
On master.

The mdb-level list-parameters call has an extra option 'searchMembers' which will attempt to match ``q`` inside top-level parameter types. Also works for searching inside array entries in case the search term includes a specific offset.

closed 2021-04-20 by fqqb