yamcs/yamcs #539: CustomAlgorithm Workflow

lorenzo-gomez-windhover, 2021-03-22:


xpromache, 2021-03-23:
We did not have aggregates for a long time. The internal subscription works at parameter level (you cannot subscribe to aggregate members), that's why the update method gets the full parameters not the members.

I have recently introduced a class ActiveAlgorithm class;
I think your code that extracts the member could be moved there in the updateParameters method. You can give it a try if you want.

lorenzo-gomez-windhover, 2021-03-23:
I see what you mean. I would love to add support for this, but at the moment it's not a priority. We should be able to do what we want to do in Studio scripts. Thanks so much for the help!

xpromache, 2021-03-24:
There is an advantage of having values computed using algorithms on the server: the results are available to all clients (web, grafana) and are also saved in the archive.

The Studio scripts are great for doing display stuff (for example a little animation or adding some colors) but that's about it.

commit 5eed9276 (2021-04-14): java-expression algorithms

Added algorithms of type "java-expressions" - these are bits of Java code compiled by Yamcs on the fly (unlike the java algorithms whcih require a jar to be installed on the server).
Also fixed #539 the inconsistencies in aggregate values/member values - now if the algorithm is declared as needing an aggregate member value as input, the updateInput in AbstractAlgorithmExecutor will be called with that member value rather than the whole aggregate.
commit 9564d780 (2021-04-20): java-expression algorithms

Added algorithms of type "java-expressions" - these are bits of Java code compiled by Yamcs on the fly (unlike the java algorithms whcih require a jar to be installed on the server).
Also fixed #539 the inconsistencies in aggregate values/member values - now if the algorithm is declared as needing an aggregate member value as input, the updateInput in AbstractAlgorithmExecutor will be called with that member value rather than the whole aggregate.
closed 2021-04-20 by xpromache