yamcs/yamcs #541: how to get 'tagTime' required for 'Update Tag'/ 'Delete Tag' API

nikhilmandlik, 2021-03-29:
I have created a tag as follows and want to edit it using Update Tag API, but not sure where can I get `tagTime` mentioned in API

## CREATE TAG:
```
POST: http://localhost:8090/api/archive/myproject/tags

BODY:
{
  "name": "red",
  "color": "#ff0000"
}
```

## List Tags:
```
GET: http://localhost:8090/api/archive/myproject/tags

Response:
{
    "tag": [
        {
            "id": 1,
            "name": "red",
            "color": "#ff0000"
        },
        {
            "id": 2,
            "name": "orange",
            "color": "#ffa500"
        },
        {
            "id": 3,
            "name": "blue",
            "color": "#0000ff"
        }
    ]
}
```

merose, 2021-03-29:
It looks like the tag time is the Yamcs instant for the start time that you get back from the Create Tag response.

Create a tag:

```
$ curl --data-ascii '{name: "gorp", start: "2021-03-29T20:00:00Z", stop: "2021-03-29T23:00:00Z", description: "some desc", color: "#FF0000"}' --header 'Content-Type: application/json' --request POST 'http://localhost:8090/api/archive/viper/tags'
{
  "id": 1,
  "name": "gorp",
  "start": "1617048037000",
  "stop": "1617058837000",
  "description": "some desc",
  "color": "#FF0000",
  "startUTC": "2021-03-29T20:00:00Z",
  "stopUTC": "2021-03-29T23:00:00Z"
}
```

The start time, as a Yamcs instant, is 1617048037000. (Note that this is the UNIX time of 2021-03-21T20:00:00Z + 37 leap seconds, the current difference between TAI and UTC.) Change the description:

```
$ curl --data-ascii '{name: "gor", start: "2021-03-29T20:00:00Z", stop: "2021-03-29T23:00:00Z", description: "another desc", color: "#FF0000"}' --header 'Content-Type: application/json' --request PATCH 'http://localhost:8090/api/archive/viper/tags/1617048037000/1'
{
  "id": 1,
  "name": "gorp",
  "start": "1617048037000",
  "stop": "1617058837000",
  "description": "another desc",
  "color": "#FF0000",
  "startUTC": "2021-03-29T20:00:00Z",
  "stopUTC": "2021-03-29T23:00:00Z"
}
```

So you'll probably have to save away the `start` value in the Create Tag response in order to manipulate the tag.

merose, 2021-03-29:
However, it's unclear why the time is required in the API if the ID is unique. (I see in the Yamcs code that both the time and the ID are used as the lookup key in RocksDB, but that seems an implementation detail that maybe should not be required in the API, unless I am misunderstanding.)

xpromache, 2021-03-30:
Indeed the tag requires knowing the start time.

But we want to get rid of those tags in favor of the timeline - in the timeline the tags will be replaced by "items" and there will be more "intelligent" items called activities which can trigger execution of things on the server. 
 
I pushed here the current snapshot of the timeline:
https://github.com/yamcs/yamcs/blob/timeline/yamcs-api/src/main/proto/yamcs/protobuf/timeline/timeline.proto
https://github.com/yamcs/yamcs/blob/timeline/docs/server-manual/services/instance/timeline.rst

I think the basic operations on items should be at the same level or better than the tags.

closed 2021-03-31 by fqqb