yamcs/yamcs #542: qualifiedNames for members of Aggregates

lorenzo-gomez-windhover, 2021-03-30:
Hi there,

hope you are all doing well.

So I was playing around with the http APIs and noticed that members of Parameters do not have a `qualifiedName` attribute like `ParameterInfo` does. However in the [documentation](https://docs.yamcs.org/yamcs-http-api/mdb/get-parameter/) it states the following:
```
interface MemberInfo {
  name: string;
  qualifiedName: string;
  shortDescription: string;
  longDescription: string;
  alias: NamedObjectId[];
  type: ParameterTypeInfo;
}
```

Is this documentation correct or am I not sending my http request correctly?

Here is  my code:
```
    response = requests.get("http://localhost:8090/api/mdb/{INSTANCE}/parameters")
    parameters = response.json()['parameters']
```

Here is what the server gives me:

```
parameters[0]['type']['member'][0]:
{'name': 'Calibration', 'type': {'engType': 'aggregate', 'member': [{'name': 'MagXScale', 'type': {'engType': 'float'}}, {'name': 'MagYScale', 'type': {'engType': 'float'}}, {'name': 'MagZScale', 'type': {'engType': 'float'}}, {'name': 'MagXOffset', 'type': {'engType': 'float'}}, {'name': 'MagYOffset', 'type': {'engType': 'float'}}, {'name': 'MagZOffset', 'type': {'engType': 'float'}}, {'name': 'MagRotation', 'type': {'engType': 'integer'}}, {'name': '_spare_end', 'type': {'engType': 'integer'}}]}
```

Am I missing something or is there a typo in documentation? 

Hope my question makes sense.

Thanks
Lorenzo

xpromache, 2021-03-30:
I think the qualifiedName in the MemberInfo is a mistake so I think we should remove it.

Are you trying to use that name for something? I think you can compose a name (for a parameter request for example) from the parameter qualified name and the member name.

lorenzo-gomez-windhover, 2021-03-30:
Yes, but I was able to assemble entire name myself by doing the following:

```
def get_aggregate_qualified_names(parameter, param_qualified_name, out_names: [str]):
    for member in parameter['type']['member']:
        if member['type']['engType'] == 'aggregate':
            get_aggregate_qualified_names(member, param_qualified_name + MEMBER_SEPARATOR + member['name'], out_names)
        else:
            out_names.append(param_qualified_name + MEMBER_SEPARATOR + member['name'])
``` 

It's not big deal, everything necessary is on the server already to assemble the name. I was just curious because the documentation was hinting at each member having its own qualifiedName, but it's pretty trivial to figure it out by drilling down on the aggregate.

xpromache, 2021-03-30:
you may need to adjust your code if the parameter is an array.

lorenzo-gomez-windhover, 2021-03-30:
Thanks, good point! But we don't use arrays(at least at the moment) :). We flatten them inside the aggregates.

lorenzo-gomez-windhover, 2021-03-30:
The reason for this actually is because at the time(if recall correctly), YAMCS did not support arrays inside of Aggregates. Are arrays supported now for Aggregates?

xpromache, 2021-03-30:
> The reason for this actually is because at the time(if I recall

I think you can have arrays of aggregates but not inside. A reason I
remember is that in XTCE it is not possible to specify the size of the
array except if the array is extracted directly in the packet (with the
Container-> ArrayParameterEntry).

But maybe fixed size arrays inside aggregates should work - it should be
possible to support this relatively easily if you need it.

xpromache, 2021-03-31:
> It would be nice to have. Any chance you guys can use `ArrayParameterType` for this kind of thing? Assuming that arrays are of a fixed size of course.

It turns out that the XTCE 1.2 arrays do specify their size in their type (even dynamic size) so I made now some changes which should allow arrays inside aggregates. 

One interesting test case which I didn't try but you are welcome to do is to have an aggregate with one member the array size and another member an array of that size.

commit 388fd04f (2021-03-31): removed qualifiedName from member info

Aggregates members are defined as part of the type so a qualified name does not make sense for them.
The property was there but never set.
Fixes #542
closed 2021-03-31 by xpromache