yamcs/yamcs #544: Could you explain the reasoning for the container change in 5.4.2

AdamElliott-AB, 2021-03-31:
I found it very useful to see the specific containers on the yamcs-web home page and in the telemetry->packet page. What was the reasoning for grouping packets by their more general containers?

From the changelog:
Changes in Yamcs 5.4.2 (March 9, 2021)
Fixed the container name used as partition in the tm table. XtceTmRecorder will only use as partition (pname column in the tm table) the containers directly inherited from the root container defined on the stream on which the TM packet is received. Previously, if a container included another container inside, this last one would have been used as partition and also appear as the name of the packet in the home page of yamcs-web.

xpromache, 2021-03-31:
Sorry, that was a bug, it is fixed on master, I knew I had to make a
release otherwise it would bite me....

I'll do it tomorrow.

xpromache, 2021-04-14:
the release 5.4.5 should fix this issue.

closed 2021-04-14 by xpromache