yamcs/yamcs #546: Database Limits

lorenzo-gomez-windhover, 2021-04-01:
Hi there,

I've using the YAMCS DB Shell and I executed the following command:

```
 select * from cmdhist 
```

And I get all of the records, however, when I send more than 2000 commands, I don't get any more records.

Is there a limit on the size of the database? If there is, can that be changed through configuration?

I looked around in the documentation and could not find anything about any limit regarding the database. Perhaps I missed it?

I hope my question makes sense.

Thanks
Lorenzo

xpromache, 2021-04-01:
No, there is no limit but that db shell is not able to return more than
2000 rows currently because we haven't implemented pagination (yet).

You can get the command history with this:
https://docs.yamcs.org/yamcs-http-api/commands/list-commands/

lorenzo-gomez-windhover, 2021-04-01:
Thanks!

So the database grows to the size of disk(however big that may be) on the server that YAMCS runs on?

xpromache, 2021-04-01:
Yes, indeed.
You should hopefully monitor the free diskspace to do something about it
before it runs out...

Note that the RocksDB database is using something called log-structured
merge-tree (https://en.wikipedia.org/wiki/Log-structured_merge-tree) and
from time to time it combines files from a lower level to a higher level so
it can suddenly grow a lot (but only temporarily ) without much data coming
in.