yamcs/yamcs #550: Use influxdb

Amr-Navy, 2021-04-12:
I want to archive the parameters name and value in Influxdb side by side with Rocksdb.
Is it simple? and from where can I start?

xpromache, 2021-04-13:
Don't know too much about influxdb so I cannot answer very precisely.

You first need to decide if you want (near) realtime data or batch import
is enough.

You also need to decide how and what to store in InfluxDB. On Yamcs each
parameter has a generation time, reception time, raw value, engineering
value and parameter status (including alarms, expiration, validity ranges).
I guess you can start with storing only generation time and engineering
value and discarding everything else.

Simplest is to make a python script to extract data from Yamcs and push it
to InfluxDB. You can use the Yamcs python client:
https://docs.yamcs.org/python-yamcs-client/

A more performant option would be to have a service on Yamcs that pushes
the data to influxdb as it would avoid the double translation. If you are
serious about it and have some java programming experience, I can help you
with that.

Amr-Navy, 2021-04-14:
I prefer the more performant option, and the generation time and parameter name engineering value is exactly what I want to store.

xpromache, 2021-04-14:
Very good then. I think you can follow these steps:
1. Make a test program to write data from Java. I see influxdb has a java client:
https://github.com/influxdata/influxdb-client-java

2. Make a yamcs service that connects to the realtime processor and gets data.
A yamcs service is soemthing that inherits from AbstractYamcsService. In the init method you would read and validate the config (see other services, there are plenty of examples for that), in the doStart method you subscribe to all parameters with something like:
```
 realtimeProcessor = YamcsServer.getServer().getProcessor(yamcsInstance, processorName);
        if (realtimeProcessor == null) {
            throw new ConfigurationException("No processor named '" + processorName);
        }
subscriptionId = realtimeProcessor.getParameterRequestManager().subscribeAll(this);
```
The subscriptionId you remember to use in the doStop method to unsubscribe.
Your service has to implement the ParameterConsumer interface and you get called via 

```
void updateItems(int subscriptionId, List<ParameterValue> items);
```
3. In the updateItems method you have practically have to send the items to influxdb. 
4. You have two methods  ParameterValue.getEngValue() and ParameterValue.getGenerationTime(). The value you have to deal differently depending on type. The time you can convert to something that influx understands using the TimeEncoding class.

5. Once you have the service, you put it in the yamcs.instance.yaml with some configuration (at least some settings for how to reach influx).

6. When you have that running, next things is to decide how to deal with old data (if the service is down for whatever reasons or data is received in non-realtime).  Probably you will want an REST API to allow users to make requests for filling up the influxdb with past data.

Amr-Navy, 2021-04-18:
Thank You it works.
[image]

xpromache, 2021-04-18:
Great! 

You want to make that public?

We also have a small plugin api which you can implement so it appears in the yamcs-web plugin section. You can look at an example here:
https://github.com/yamcs/yamcs-prometheus/blob/master/src/main/java/org/yamcs/prometheus/PrometheusPlugin.java

Then you can also register some REST endpoints for user to call, for example to resend data to influxdb in case it changes on Yamcs via a non-realtime stream.

And a nice getting started documentation... and the humanity will remember you forever :)

Amr-Navy, 2021-04-20:
Sure I will do my best but I think I will need some time and your help :)

xpromache, 2021-04-22:
> Sure I will do my best but I think I will need some time and your help :)

Let me know when there is something to review.

Amr-Navy, 2021-04-28:
can you check this https://github.com/Amr-Navy/Yamcs-Influxdb

xpromache, 2021-05-04:
Hi, I had a quick look and have a few comments:

1. Do you mind using this
https://github.com/yamcs/yamcs/blob/master/eclipse-yamcs-style.xml for
formatting the code?
  In particular, package names should be in lowercase, method names should
start with lowercase, curly brackets on the same line with the method,
indenting...

2. This code
   Instant.ofEpochMilli(pv.getGenerationTime()).minusMillis(37000);
should be replaced with something else after deciding what to do with the
leap seconds.
If InfluxDB doesn't know about them I think the best strategy is to use
smearing like we do when sending data to external clients.
You can use this method
https://yamcs.org/javadoc/yamcs/org/yamcs/utils/TimeEncoding.html#toProtobufTimestamp(long)
for the time being.

3. The aggregate values should also be sent, I don't know how influxdb
works with aggregates.

4. Maybe we can also send the parameter status using influxdb tags.

5. Do you accept pull requests?

Amr-Navy, 2021-05-05:
I have solved all these issues and update the repository u can check.
for sure i will accept your pull requests  @xpromache

closed 2023-07-05 by fqqb