yamcs/yamcs #555: UdpTcFrameLink cannot be disabled at startup

merose, 2021-04-21:
To reproduce:
1. Configure a data link of class `org.yamcs.tctm.ccsds.UdpTcFrameLink` and specify `enabledAtStartup: false` in the configuration.
2. Start Yamcs.
3. Go to the _Data Links_ menu option.

Expected result: UDP frame data link and all child virtual channels are disabled.
Actual result: UDP frame link and all children are enabled.

Curiously, this works correctly for other data link types. It also seems that `UdpTcFrameLink` might be able to derive from `AbstractThreadedTcDataLink` to reduce code duplication, as `UdpTcDataLink` does.

closed 2021-04-22 by xpromache

xpromache, 2021-04-22:
Fixed on master (the reason for the failure was a NPE in the doDisable() method, contrary to what I initially thought that is the getConnectionStatus() method).

Note that the child links are not automatically disabled but I added another method isEffectivelyDisabled which returns true if the link or its parent is disabled. The LinkManager will use the new method to insert a negative Sent ACK in the command history instead of leaving the command hanging,

I don't think I can use the AbstractThreadedTcDataLink because it only works for simple links not aggregated links. 
The simple links have a command queue and a rate limiter.
The aggregated frame link have a multiplexer which selects a command frame from one of the sublinks based on a priority scheme.
When the multiplexer wants to get a command frame from one sublink (i.e. a virtual channel), then the sublink can build the frame from multiple commands in its queue depending on the maximum frame size.

merose, 2021-04-22:
That makes sense. I did not look in detail at whether it would be possible, but saw some methods in common. Thanks!

commit 94456087 (2021-04-22): fix for link disable

- fix the UdpTcFrameLink disable status (although the link was disabled, it was still showing ok). Fixes #555
- when the UdpTcFrameLink was disabled and then stopped, a NPE was generated - fixed that
- added a method isEffectivelyDisabled which will return false for sub-links if the parents are disabled even if the link itself is enabled. That will cause the LinkManager to issue a negative Sent ack rather than leave the command hanging