yamcs/yamcs #559: Parameter archive filler does not properly handle aggregate types

merose, 2021-05-03:
Found using Yamcs quickstart in Yamcs 5.4.5.

The Yamcs backfiller and realtime filler do not properly write aggregate values to the parameter archive. (Or it might be a retrieval problem, not yet sure.) This can be seen by defining two identical SequenceContainers, one containing three float parameters, the other containing a single parameter that is an aggregate of 3 floats.

The tar-ball in the instructions below modifies the quickstart project in these ways:
- Changes Yamcs version to 5.4.5
- Adds another XTCE file, Test.xml
- Adds an XTCE subloader to load Test.xml
- Reduces the backfiller fill frequency to 60 seconds, so that we do not have to wait 10 minutes for the archive fill to occur
- Adds a Python script, `send_rover_pose.py`, to the quickstart root directory.

The Test.xml XTCE file defines a 32-bit floating point parameter type and an aggregate `Vector3` of three of those types. The aggregate has members `x`, `y`, and `z` to hold a position. There are four parameters defined, `roverX`, `roverY`, `roverZ`, and `roverPose`, of the aggregate type.

There are two SequenceContainers with APIDs 101 and 102. The first contains the three individual parameters, the second contains only the `roverPose` aggregate. If the parameter archive works identically for individual parameters and aggregates, then sending to those two APIDs should allow us to test whether the parameters are written to the archive in the same way. However, as will be seen, only the individual parameters are archived. (Unless the problem is in archive retrieval, of course.)

# To reproduce

1. Clone the Yamcs quickstart project. (git@github.com:yamcs/quickstart.git)
2. In the quickstart root directory, unpack this tar file via `tar xf test.tar.gz`: [test.tar.gz](https://github.com/yamcs/yamcs/files/6418247/test.tar.gz)
3. Start Yamcs: `$ mvn yamcs:run`
4. In another terminal window, send the two packets:
```
$ python send_rover_pose.py --apid 102 10 20 30
$ python send_rover_pose.py --apid 101 10 20 30
```
5. Verify that both an individual parameter and the aggregate parameter have a single value:
```
$ curl 'http://localhost:8090/api/archiv/myproject/parameters/myproject/Test/roverX'
$ curl 'http://localhost:8090/api/archiv/myproject/parameters/myproject/Test/roverPose'
```
6. Wait one minute for the backfiller to run: `$ sleep 60`
6. Stop Yamcs with Ctrl-C.
7. Start Yamcs again: `$ mvn yamcs:run`
8. Check that both the individual parameter and the aggregate parameter have a single value in the archive:
```
$ curl 'http://localhost:8090/api/archiv/myproject/parameters/myproject/Test/roverX'
$ curl 'http://localhost:8090/api/archiv/myproject/parameters/myproject/Test/roverPose'
```

**Expected result:** Both `roverX` and `roverPose` have a single value in the archive.

**Actual result:** Only `roverX` has a value in the archive. The archive shows zero values for `roverPose`.

# Additional info

- Querying for a single member of the aggregate, such as `roverPose.x`, also fails in the same way.
- Enabling the realtime filler and disabling the backfiller also fails in the same way.

xpromache, 2021-05-05:
The test data which you provided doesn't work - the xtce-header is not there and the CCSDS and TelemetryPacket aren't defined.
After copying some containers around I manged to start and I do notice the behaviour you notice except for the last bit:  Querying for a single member of the aggregate, such as roverPose.x, also fails in the same way.

Querying a member aggregate does work for me:

 curl 'http://localhost:8090/api/archive/myproject/parameters/myproject/Test/roverPose.x'
```
{
  "parameter": [{
    "id": {
      "name": "roverPose.x",
      "namespace": "/myproject/Test"
    },
    "rawValue": {
      "type": "DOUBLE",
      "doubleValue": 10.0
    },
    "engValue": {
      "type": "DOUBLE",
      "doubleValue": 10.0
    },
    "generationTime": "2021-05-05T02:31:32.923Z",
    "acquisitionStatus": "ACQUIRED",
    "processingStatus": true,
    "generationTimeUTC": "2021-05-05T02:31:32.923Z"
  }]
}
```

Note that as soon as I run again the script to inject new data, I see back the values for the whole aggregate, including the old values.

Problem seems to be with reconstructing the aggregate value from members - if it doesn't have any current value it seems not to know it has to deal with an aggregate. Currently the parameter archive has no memory that a parameter named "roverPose" exists, it only stores the individual aggregate values with their names "roverPose.x", "roverPose.y" and "roverPose.z".

Probably I have to change that;
The design was such that one can change the type of a parameter without changing its name and the archive shall cope with it.  

For that, it allocates internally a parameter id (uint32) for each (parameter qualified name, raw type, eng type) combo  and all the data is stored with that id.
Currently the aggregate parameters are not allocated an id (but each member has one) so if a new scalar parameter comes with the same name as the aggregate, the new parameter will get an id shadowing somehow the aggregate. 

I think I need to allocate an id to the aggregate and store it together with the ids of its members. 
to be continued...

xpromache, 2021-05-05:
I see some other oversight when working with aggregates; these queries produces a 500 internal server error:
http://localhost:8090/api/archive/myproject/parameters/myproject/Test/roverPose/samples
http://localhost:8090/api/archive/myproject/parameters/myproject/Test/roverPose/ranges

They should produce probably a 400 bad request

merose, 2021-05-05:
Sorry I wasn't clear. My intention was to have the _Test_ SpaceSystem loaded as a sub-system of the quickstart _myproject_, and to inherit the CCSDS definitions from `xtce.xml` from quickstart. Hmmm... I'll try the `roverPose.x` queries again.

merose, 2021-05-05:
Sorry I wasn't clear. My intention was to have the _Test_ SpaceSystem loaded as a sub-system of the quickstart _myproject_, and to inherit the CCSDS definitions from `xtce.xml` from quickstart. Hmmm... I'll try the `roverPose.x` queries again.

(Perhaps in the future I'll just include a patch file to modify the XTCE in quickstart, rather than adding a new file, to make the setup clearer.)

commit 03c4750e (2021-06-03): Refactor extraction from the parameter archive

- recompose aggregate and arrays out of components. Fixes #559
- created several iterators to simplify the extraction processor
closed 2021-06-03 by xpromache