yamcs/yamcs #573: Auto-Stack / periodically sending commands

nmaas87, 2021-06-18:
Dear all,

would there be a possibility to allow an existing Stack in the webinterface to be sent every x minutes until its canceled?
(Baiscally like a cronjob on Stack base, would probably need some job list showing when next execution would be and to cancel the exceution)

Kind regards

Nico

fqqb, 2021-06-21:
The current situation is that stacks and stack advancement are fully client-controlled (be it via yamcs-web or yamcs-studio).

We are interested in adding automated stacks and/or procedures as part of our “Yamcs Timeline” functionality that we have been preparing. If you’re interested in helping this move faster, feel free to reach out to us for commercial work.

With existing tools, probably sending commands via a Python script would be the best approach.

nmaas87, 2021-06-21:
Thank you @fqqb - thats good to hear about the Yamcs Timeline, we are looking forward to that and will consider the Python script :)!

closed 2021-06-21 by nmaas87