yamcs/yamcs #576: Impossible to write unit tests for custom algorithm because of AlgorithmExecutionResult member visibility

merose, 2021-07-02:
`AlgorithmExecutionResult` has three member variables:

---
    final List<RawEngValue> inputValues;
    final List<ParameterValue> outputValues;
    final Object returnValue;
---

However, they are all default scope, and there are no getters. It's very difficult to write a unit test for a custom algorithm, because the algorithm's `execute` method returns `AlgorithmExecutionResult`, and the unit test cannot access the members to perform verifications.

Either add getters for those members (preferred, I think) or change the scope to `public`.

(This was not a problem prior to Yamcs 5.5.x, because algorithms returned a parameter value list instead.)

Workaround: Change the algorithm to return a class derived from `AlgorithmExecutionResult` _that includes getters for the members_, and change the unit test to cast the result to the derived class in order to gain access.

merose, 2021-07-02:
Workaround: Create an `innerExecute` method in your algorithm that returns a `List<ParameterValue>` like the old `runAlgorithm` used to return, and call that from unit tests instead. The `execute` method can then just `return new AlgorithmResult(innerExecute(...))`.

fqqb, 2021-09-04:
On master (added getters).

closed 2021-09-04 by fqqb