yamcs/yamcs #577: Sheet Loader does not resolve members of aggregate parameters referenced in conditions

isarantidis, 2021-07-06:
When using the import from Excel sheet feature and specifying a container with a condition such as:

/MISSIONDB/CCSDS_Header/APID==1

where CCSDS_Header is a Parameter of type CCSDS_Header_Type, which is an aggregate parameter with one of the members being APID, I receive the following error:

_Failure while attempting to validate configuration[0;0m: java.lang.IllegalArgumentException: Reference to an aggregate parameter type CCSDS_Header_Type without speciyfing the path
	org.yamcs.xtce.Comparison.validateValueType(Comparison.java:69)
	org.yamcs.xtce.ConditionParser.lambda$toComparison$2(ConditionParser.java:239)
	org.yamcs.xtce.util.ParameterReference.resolved(ParameterReference.java:39)
	org.yamcs.xtce.util.ReferenceFinder$FoundReference.resolved(ReferenceFinder.java:349)_

It seems that org.yamcs.xtce.ConditionParser#toCondition adds a generic ResolvedAction instead of the more specific ParameterResolvedAction, losing member path information

xpromache, 2021-07-06:
I will look into this later, can you please try with . (dot) separator?

/MISSIONDB/CCSDS_Header.APID==1

isarantidis, 2021-07-06:
I tried and got the following error:

_Cannot parse parameter for comparison '/MISSIONDB/CCSDS_Header.APID'. Use parameterName or parameterName.raw_

Tried also with:

-  /MISSIONDB/CCSDS_Header.APID.raw and got the same error
-  /MISSIONDB/CCSDS_Header/APID.raw but got the initial error again

xpromache, 2021-07-06:
Indeed, it looks like a bug.

We never used container inheritance with conditions on aggregate
members, this is something I've seen in some XTCE files and never tested it
with the spreadsheet loader.

I will solve it and let you know.