yamcs/yamcs #580: Command fields with <initialValue> are hidden by default in the browser GUI

tardyantoine, 2021-07-09:
Hi all,

If the <initialValue> tag is used to prefill a command field, that field is by default hidden in the browser commanding GUI. The operator needs to click the arrow with a "Show n arguments with defaults" label to be able to set them (see screenshot).

I believe this is not a totally safe practice. If a command field needs to be fixed, then it should not be displayed at all or not updatable (as is already the case). If a command field can be updated by the operator, then hiding it away presents a risk for human error, as the operator might not see or skip the defaulted fields.

Would it be possible to add a tag that prevents the defaulted values to be hidden in the commanding GUI?

[image]

fqqb, 2021-07-12:
Sounds OK. I'll add an opt-in to get rid of the show/hide toggle.

fqqb, 2021-07-29:
On master.
From next release, you will be able to disable this collapsing in yamcs.yaml (default is true):

    yamcs-web:
      collapseInitializedArguments: false

closed 2021-07-29 by fqqb