yamcs/yamcs #584: How to track scheduled ground station passes/contact planning?

YeomansIII, 2021-07-21:
### Discussed in https://github.com/yamcs/yamcs/discussions/583

<div type='discussions-op-text'>

<sup>Originally posted by **YeomansIII** July 21, 2021</sup>
I am trying to determine if Yamcs has a way to schedule and track ground station passes/contacts, triggering scripts/commands during a pass. I have been able to schedule an "event" in the future, but an event does not have a start and end time, only a reception/generation time.

Could someone provide details on how they schedule and track their ground station passes with Yamcs?</div>

merose, 2021-07-21:
We are planning on making this a manual operation, but facilitated by a shell script to make the handover between ground stations easier. One could easily schedule such a script instead. The Python API for managing data links programmatically is here: https://docs.yamcs.org/python-yamcs-client/link/.

xpromache, 2021-07-22:
Correct, currently the solution would be to implement some external scripts using for example the python client:
https://docs.yamcs.org/python-yamcs-client/

We have in development something called Yamcs Timeline which would allow scheduling this kind of scripts by the Yamcs server itself and also have a user interface part of the Yamcs web. However I'm not sure when this will be available, I was hopping this summer but chances are low...

YeomansIII, 2021-07-22:
Thanks for the replies. Has the development of the yamcs-timeline feature started? Or is it in planning? Depending on circumstances, I may be able to assist with Java and Angular development.

fqqb, 2021-07-22:
Hi, next release will include a first iteration of where we want to go with that functionality. This will mostly focus on visuals and basic interactions only, and in particular will not (yet) include any scheduling support.
You’re welcome to have a look at it then.