yamcs/yamcs #585: Is it possible to keep two Yamcs servers in sync for a given instance?

keck-in-space, 2021-07-21:
We have two servers setup to run Yamcs.

Server One (target_server) is is communicating directly with hardware and has incoming data. 
Server Two (analysis_server) is setup at a different location and has no data.

We would like to synchronize the data for a given instance between Server One and Server Two. Is there a supported way to do this?

Our current approach is to utilize the HTTP API and request the data as it is added to Server One, however this is fraught with various issues. If we lose the connection, we could lose parameter data, and it may be difficult to determine exactly what data is missing on Server Two.

Any ideas for a better approach here?

Thank you.

merose, 2021-07-21:
There is a replication service. We found it a little tricky to set up, but quite workable. There are configuration examples in yamcs/examples/, and there is documentation about the master and slave service instance at https://docs.yamcs.org/yamcs-server-manual/services/instance/replication-master/ and https://docs.yamcs.org/yamcs-server-manual/services/instance/replication-slave/.

keck-in-space, 2021-07-21:
Very interesting. This looks like it could work. Does this work across multiple Yamcs servers that are separated by the Internet? I only ask because the following wording in the docs was confusing.

> "Multiple master and slaves from different Yamcs instances in the same Yamcs server will register to the same replication server."

Taken from here: https://docs.yamcs.org/yamcs-server-manual/services/global/replication-server/

merose, 2021-07-21:
Yes, it does work across machine boundaries. The example configurations in yamcs/yamcs/examples/ replicate between instances on the same server, but that is not required. We have tested the replication between instances on two different Yamcs servers.

keck-in-space, 2021-07-21:
This solves my issue. Thanks so much!

closed 2021-07-21 by keck-in-space

xpromache, 2021-07-22:
SSL (encryption) should also work although it's not part of the examples.