yamcs/yamcs #587: HTTP API Issue Command with parameters

keck-in-space, 2021-07-22:
Hello,

I am looking at the HTTP API and I need some help understanding the [Issue Command](https://docs.yamcs.org/yamcs-http-api/commands/issue-command/) call. How could I specify the parameters of the command with this call? I only see a way to state the command name.

Thank you!

fqqb, 2021-07-22:
You can assign argument using the JSON request body:

    {"assignment": [{"name": "arg1", "value": "value1"}, {"name": "arg2", "value": "value2"}]}

The web page specifies the format of the request body in TypeScript format. If you intend to do more with HTTP API, it can also be useful to use your browser development tools and snoop on the calls that the Web UI does.

keck-in-space, 2021-07-22:
Ah, I see it now. Thanks! If I have a binary argument is it supported somehow? I don't think JSON supports binary natively.

fqqb, 2021-07-22:
It's strings only. You should pass as hex.

keck-in-space, 2021-07-22:
OK. Great. Thank you. I will try this.

closed 2021-07-22 by keck-in-space