yamcs/yamcs #591: Stopping replays

nmaas87, 2021-08-06:
There are multiple options to leave a replay both in Web and Studio - but I have not found an option to actually delete/stop a replay in progress. Did I miss something?

Best regards,

Nico

fqqb, 2021-08-09:
This should indeed be added.

In the old days (before yamcs-web) replay processors were started in non-persistent mode. These are processors that stop immediately when no more clients are connected. But that didn't work well in a web-context (refresh page => replay gone), so we made replays persistent, but then never got to adding UI to get rid of them (other than a server restart).

fqqb, 2022-03-02:
As of v5.6.0 you have an option "Leave & close replay" available.

closed 2022-03-02 by fqqb