yamcs/yamcs #592: Multi-parameter extraction in web

nmaas87, 2021-08-06:
Dear all,

within the Webinterface its possible under Telemetry -> Parameters -> (choose parameter) -> Historical data to extract the data for this given parameter through a timespan. I would like to see two improvements:
a) Please include the uncalibrated version of the parameter within the extraction like in the Python API (list_parameter_values)
b) Please allow users to generate "parameter-lists" out of the existing parameters of e.g. the MDB. And then allow for these parameter-lists to be extracted inculding the calibrated and uncalibrated versions of the configured item over the set timespan by the user.

This would really help out, especially if OPS people need to extract data on a regular base.

Thanks a lot,

Nico

fqqb, 2021-08-09:
Hi for (b), there is already a roundabout way to get what you want:
* Make a Display of type 'Parameter Table'.
* Add the parameters of interest to it.
* From the top menu, choose 'Export archive data', and you will get a dialog with a time range of your choice.

These are a lot of steps, so I agree we should have something more intuitive.

nmaas87, 2021-08-09:
Thank you for the initial feedback, that works - with the limitation of not having the uncalibrated values within the download - as you already wrote :). Would be really good to make it more easy and with a) - but thanks already, had not found this specific Export function by searching until your post 👍🏻