yamcs/yamcs #604: Multidimensional Arrays

lorenzo-gomez-windhover, 2021-09-16:
Hi there,

hope you are all doing well.

Imagine a multidimensional array such as the following:
```
			<ArrayParameterType name="matrix2D" arrayTypeRef="/dt/float32">
			  <DimensionList>
			    <Dimension>
			      <StartingIndex>
			        <FixedValue>0</FixedValue>
			      </StartingIndex>
			      <EndingIndex>
			        <FixedValue>2</FixedValue>
			      </EndingIndex>
			    </Dimension>
			    <Dimension>
			      <StartingIndex>
			        <FixedValue>0</FixedValue>
			      </StartingIndex>
			      <EndingIndex>
			        <FixedValue>2</FixedValue>
			      </EndingIndex>
			    </Dimension>
			  </DimensionList>
			</ArrayParameterType>
```

I know that clients are able to with no problem can access the array via the API with something like the following:

```
response = requests.get(f"http://localhost:8090/api/processors/{INSTANCE}/realtime/parameters/SIMULATOR/matrix2D[0][0]")
print(f'2D item-->{response.json()}')
```
Awesome, I can access items in multidimensional arrays.

However, when I try this it does not work as I had hoped it would:

```
response = requests.get(f"http://localhost:8090/api/processors/{INSTANCE}/realtime/parameters/SIMULATOR/matrix2D[0]")
print(f'2D item-->{response.json()}')
```

The server does not respond with the first row in the matrix, it gives me an error:

```
{'code': 404, 'type': 'NotFoundException', 'msg': 'Nonexistent array/aggregate path in name SIMULATOR/matrix2D[0]'}

```
Is this intended behavior?

Hope this issue was clear enough.

Thanks
Lorenzo

xpromache, 2021-09-17:
This functionality is not yet implemented.
Can be done, when time allows...

lorenzo-gomez-windhover, 2021-09-17:
Thanks for the quick response!
Looking forward for when this feature gets added!