yamcs/yamcs #608: Authentication with processes run by ProcessRunner

merose, 2021-09-22:
I would like to run Yamcs using authentication, and I'd like to run a program using ProcessRunner that will periodically create events in Yamcs using the Python API. However, it's unclear how that external program should authenticate. I would prefer not to embed authentication arguments in the command line configured in Yamcs. What is the best way to authenticate from a process spawned by ProcessRunner?

merose, 2021-09-22:
Or is there a way to detect that the connection request is from the same machine, and assume default authentication credentials in that case?

fqqb, 2021-09-23:
It seems that the external program could have some self-managed config on how to connect (?)

To avoid passwords, you could also delegate auth to a reverse proxy, like nginx or apache with RemoteUserAuthModule.
I've used this before to support things like certificate-based authentication from client scripts.
(see https://docs.yamcs.org/yamcs-server-manual/security/authmodules/remote-user/).

ProcessRunner could also be improved to allow setting custom environment variables. At least then they would not appear on the command line.

fqqb, 2021-09-23:
> Or is there a way to detect that the connection request is from the same machine, and assume default authentication credentials in that case?

I think a custom AuthModule could do that (cfr. RemoteUserAuthModule for ideas).

merose, 2022-11-11:
Closing this as the suggestions from @fqqb seem sufficient.

closed 2022-11-11 by merose